geolink/opentracker #1: Update the PCB fiels and schematics

AnandSingh, 2015-04-20:
Hi 

Please update the PCB files and schematics for Opentracker v2 board.

In v2 board the J2 connector is of 20 pins while in schematic and PCB it's showing 22 pins. Please update it's very tough to track the pinouts. I am looking for an uart (EXT_TX , EXT_RX) pin from J2.

Also can you update the old github  https://github.com/TIGAL/OpenTracker repository README for new location of github repository.

Thanks
Anand

AnandSingh, 2015-04-21:
HI 
Can anyone help me to setup the EXT_TXD2 , EXT_RXD2. 

While using Serial3 for EXT_TXD2, EXT_RXD2 i am only able to receive, but transmit EXT_TXD2 is not working. When I check the voltage level at EXT_TXD2 it's showing ~.97V. 

Do I need to connect IO_TO_EXT to 3.3V with EXT_TXD2. Please suggest or help me in this regard.

Thanks
Anand

m4rkw, 2015-07-23:
Hi @AnandSingh I wish I could help but I think you'll need the geolink guys to help you with this.  I think they're swamped with emails but hopefully they'll be able to get back to you at some point.

ppescher, 2015-12-14:
Design files are up-to-date with HW revision 2.2

closed 2015-12-14 by ppescher