geolink/opentracker #3: rooming costs

haegar66, 2015-05-03:
I live near to a country border.
The cost of GPRS are very expensive by international rooming.
I want to turn off the default rooming of the data connection and activate only with an SMS command again (eg in case of theft).
Can this be installed?
greeting
Claus

m4rkw, 2015-07-08:
It should be possible, but probably not easy to implement.  I think what you probably want to do is save the network code for your home network and then detect when it changes, but that's going to involve additional queries to the modem in the main loop.  Also difficult to test as you'd have to roam in order to check that it works.  I'll add it to my someday maybe list :)

m4rkw, 2015-07-18:
@haegar66 Hi, I've just added two new SMS commands which allow you to enable and disable the sending of the data packets.  These are "dataoff" and "dataon".  In both cases it will confirm that the change has been made via sms.  While in "dataoff" mode it will still poll for its location via GPS so you can still use the sms command "locate" to see where the tracker is.  Also be sure to set GSM_SEND_FAILURES_REBOOT to 0 therwise a reboot may occur which will re-enable the data sending.  Also note that if the device gets power cycled for any reason this will also occur.

Hope this helps :)  Mark

haegar66, 2015-08-01:
Hi,

I have found the solution in theory.

AT+CREG Network Registration
Parameter
0 Disable network registration unsolicited result code
1 Enable network registration unsolicited result code +CREG:
2 Enable network registration unsolicited result code with location 
information
0 Not registered, ME is not currently searching a new operator to 
register to
1 Registered, home network
2 Not registered, but ME is currently searching a new operator to 
register to
3 Registration denied
4 Unknown
5 Registered, roaming

so it could be implemitiert:

tracker.h

   int ROAMING_ON = 0;   //Roaming 0=off 1=on

#define E-Plus_de
//  #define Vodafone_de
//  #define t-mobile_de
//  #define O2_de
//  #define Orange_at
//  #define A1_Mobilkom_Austria_at
//  #define Drei_at
//  #define t-mobile_at
//  #define Bob_at

#ifdef Drei_at
   #define DEFAULT_APN "orange.web"  //default APN
   #define DEFAULT_USER "drei.at"  //default APN user
   #define DEFAULT_PASS ""  //default APN pass
#endif

#ifdef Orange_at
   #define DEFAULT_APN "orange.web"  //default APN
   #define DEFAULT_USER "web"  //default APN user
   #define DEFAULT_PASS "web"  //default APN pass
#endif

#ifdef A1_Mobilkom_Austria_at
   #define DEFAULT_APN "orange.web"  //default APN
   #define DEFAULT_USER "web"  //default APN user
   #define DEFAULT_PASS "web"  //default APN pass
#endif

#ifdef t-mobile_at
   #define DEFAULT_APN "gprsinternet"  //default APN
   #define DEFAULT_USER "t-mobile"  //default APN user
   #define DEFAULT_PASS "tm"  //default APN pass
#endif

#ifdef Bob_at
   #define DEFAULT_APN "bob.at"  //default APN
   #define DEFAULT_USER "data@bob.at"  //default APN user
   #define DEFAULT_PASS "ppp"  //default APN pass
#endif

#ifdef Vodafone_de
   #define DEFAULT_APN "web.vodafone.de"  //default APN
   #define DEFAULT_USER "vodafone"  //default APN user
   #define DEFAULT_PASS "vodafone"  //default APN pass
#endif

#ifdef t-mobile_de
   #define DEFAULT_APN "internet.t-mobile"  //default APN
   #define DEFAULT_USER "tm"  //default APN user
   #define DEFAULT_PASS "tm"  //default APN pass
#endif

#ifdef E-Plus_de
   #define DEFAULT_APN "internet.eplus.de"  //default APN
   #define DEFAULT_USER "eplus"  //default APN user
   #define DEFAULT_PASS "eplus"  //default APN pass
#endif

#ifdef O2_de
   #define DEFAULT_APN "wap.viaginterkorn.de"  //default APN
   #define DEFAULT_USER ""  //default APN user
   #define DEFAULT_PASS ""  //default APN pass
#endif

gsm.ino

void gsm_status_read()
{
   // gsm status

   // 0 Disable network registration unsolicited result code
   // 1 Enable network registration unsolicited result code +CREG: <stat>
   // 2 Enable network registration unsolicited result code with 
location information

   // 0 Not registered, ME is not currently searching a new operator to 
register to
   // 1 Registered, home network
   // 2 Not registered, but ME is currently searching a new operator to 
register to
   // 3 Registration denied
   // 4 Unknown
   // 5 Registered, roaming

   char *status_str;
   int auswahl = 0;

   char *tmp1;
   char *tmp1val;

   debug_print(F("gsm_status() started."));

   //todo check
   gsm_port.print("AT+CREG?");
   gsm_port.print("\r");

   delay(500);
   gsm_get_reply(0);

   //todo check if everything is delivered
   tmp1 = strstr(modem_reply, "AT+CREG: ");
   tmp1 += strlen("+CREG: ");
   tmp1val = strtok(tmp1, "\r");

   //checking how many bytes NON-acked
   status_str = strtok_r(tmp1val, ", ", &tmp1val);
   auswahl = atoi(status_str);

   switch (auswahl) {
     case 0:
       debug_print(F("Disable network registration unsolicited result 
code"));
       break;
     case 1:
       debug_print(F("Enable network registration unsolicited result 
code +CREG: <stat>"));
       break;
     case 2:
       debug_print(F("Enable network registration unsolicited result 
code with location information"));
     default:
       debug_print(F("ERROR"));
   };

   status_str = strtok_r(NULL, ", ", &tmp1val);
   auswahl = atoi(status_str);
   switch (auswahl) {
     case 0:
       debug_print(F("Not registered, ME is not currently searching a 
new operator to register to"));
       break;
     case 1:
       debug_print(F("Registered, home network"));
       gsm_status.at_home = 1;
       gsm_status.at_roaming = 0;
       break;
     case 2:
       debug_print(F("Not registered, but ME is currently searching a 
new operator to register to"));
     case 3:
       debug_print(F("Registration denied"));
     case 4:
       debug_print(F("Unknown"));
     case 5:
       debug_print(F("Registered, roaming"));
       gsm_status.at_home = 0;
       gsm_status.at_roaming = 1;
     default:
       debug_print(F("ERROR"));
   };

   debug_print(F("gsm_validate_tcp() completed."));

}

sms.ino
tmp = strstr(cmd, "roaming");
      if (tmp!=NULL)
          {
            debug_print(F("sms_cmd_run(): Roaming command detected"));
            if (!ROAMING_ON)
            {
               sms_send_msg("Roaming ON", phone);
               SEND_DATA = 1;
               ROAMING_ON = 1;
            }
            else
             {  sms_send_msg("Roaming OFF", phone);
               SEND_DATA = 1;
               ROAMING_ON = 0;
             }
          }

opentracker_3_0_1.ino

   //gsm status
struct gsm_status_struct {
   int at_home;   //flag for Registered, home network
   int at_roaming; //Flag for Registered, roaming
};

gsm_status_struct gsm_status;

  debug_print(F("Current:"));
    debug_print(data_current);

  gsm_status_read(); // Roaming?

if ((SEND_DATA & gsm_status.at_home ) || (SEND_DATA & 
gsm_status.at_roaming & ROAMING_ON)) {
        int i = gsm_send_data();

        if(i != 1)
        {
       //current data not sent, save to sd card
           debug_print(F("Can not send data, saving to flash memory"));
/*
        #ifdef STORAGE
           storage_save_current();   //in case this fails - data is lost
        #endif

*/

Claus Dickert

dimetron, 2015-08-12:
Hi, 

Is anyone going to merge above roaming to master or I can do it?

I suggest also to have different GPS timeouts to be set in roaming and potentially 
use RAW TCP with compression of several points using polylinealgorithm  https://developers.google.com/maps/documentation/utilities/polylinealgorithm?hl=en

agraciabona, 2016-01-26:
Hi,
I have the opposite problem than haegar66. 
I want my tracker contiues working even out of my country but actually does not.
Maybe it is because I modified tracker.h as follows:

#define DEFAULT_APN "orangeworld"
#define DEFAULT_USER "orange"
#define DEFAULT_PASS "orange"

I made this because it is the APN I want to use and my SIM card does not allow SMS (it is cheaper!).

My question is: if I use default settings:

#define DEFAULT_APN "internet"
#define DEFAULT_USER ""
#define DEFAULT_PASS ""

my tracker will connect in every coutry?

I am far away from any border and it's difficult for my to test.
I'll be happy if someone can answer me. 

Thanks
Alfredo

bullshit, 2016-01-27:
Hi Alfredo,

if think that your mobile operator blocks GPRS connections outside your country. The APN connections are always the same. This issue treated the possibility to turn off GPRS communication in foreign countries because GPRS roaming is mostly expensive.
>I want to turn off the default roaming of the data connection and activate only with an SMS command again

m4rkw, 2016-01-27:
FWIW I had the same issue using giffgaff in the uk.  Drove to france, tracker mysteriously stopped working.  I think I checked that roaming was enabled on my account before we went, but not 100% sure.  Might go back again sometime this year so if I do I'll check it again and try to figure out why it wasn't working.

agraciabona, 2016-01-27:
I think APN are not allways the same, in Spain APN for Orange is "orangeworld" in France it is "orange". When my vehicle went to France it stopped connecting. I have roaming activated in my SIM card. 
I dont know how mobile phones do to switch APN when they travel to other country. I would like to do the same or maybe there is a generic APN which works in every country... if someone can help me...

m4rkw, 2016-01-27:
Maybe there's a mechanism for pushing out APN updates when you roam, that mobile phones are programmed to handle but opentracker is not.  Just a theory.

agraciabona, 2016-01-29:
Please haegar66, maybe you can help me: which APN do you use that provokes your opentracker roam?

agraciabona, 2016-05-13:
Hi,
I want my tracker roams but it doesnt: when it gets out of Spain it stops transmiting. 
Anyone can tell which APN should be used to make it roam???
Thanks

ZdenekAster, 2016-05-13:
In Czech Republic GSM operators, APN is some at roaming. But some operators in roaming, send html page with text and button to accept roaming data connect.... Without this click to button roaming not working :-(

m4rkw, 2016-05-14:
@agraciabona I haven't managed to get mine working with roaming yet either, probably because like you say the APN changes.  I think with the current software the only way to get it to work would be to put the sim card into a modern mobile phone, see what the APN gets automatically set to and then configured that in the tracker.

Daniel-dk, 2016-07-18:
you could look at services like Jasper wireles or telit - They are "international network providers 

 - they are a bit more expensive than regular operators but have the same APN and manage the roaming for you so the device does not need separate settings per country
We've used Telit before and it worked great ( sub-Saharan Africa )- They need to know how many devices and which countries you want to roam in to give you pricing, so it a bit of a pain, but can provide up to 3 test SIMS.

 - [JASPER wireless ( now Cisco )](https://www.jasper.com/)
 - [Telit ( I think they use Jasper in the background )](http://www.telit.com/products-and-services/iot-connectivity/sim-cards-data-plans/)