geolink/opentracker #13: imei=D

Fyod, 2015-06-26:
Hi!
I'm testing this library with my Quectel device and unfortunately I always get imei=D.
If I write a program that only shows the imei in the loop, it shows the correct imei every time.
What could be wrong?

Thanks!

Fyod, 2015-06-26:
Dubug:

gsm_get_imei() started
Modem Reply:
AT+GSN

Modem Reply:
865733020106159

Modem Reply:

Modem Reply:
OK

gsm_get_imei() result:
*
gsm_get_imei() completed
setup() completed

TonyArmitstead, 2015-06-29:
IMO the code is currently broken for reading time/date and IMEI from the modem. I had exactly the same problem as you did, but this was just one of many problems the code has. I am currently working through the code re-writing and fixing what _I_ perceive as defects. I currently have a 'working' unit that is reporting position to geolink.io but there are still many other problems to fix and other features _I_ want to add. I am not ready yet to share my changes but can guide you on how to fix the above problem if you are willing to make changes to the code you have.

Neville77, 2015-06-29:
I also have a device which polls quite well to geolink.io but we are trying  to get it send the data to our own server. We are getting a connection to our server but are struggling to get it to send the actual data string to the server. I am interested  in your work  Tony as we wish to add more features  to this if it is reliable code, and we are able to aquire more hardware.

TonyArmitstead, 2015-06-29:
Yes - I also had this problem as well. I changed the code to build a complete HTTP POST message as a single buffer and send it as a single transaction - instead of using the packet chunking code that was already there. This now sends the data fine but there are clear issues of max packet length (which is ok for now) and there are still some timeouts waiting for responses from the modem - still to fix those.

Fyod, 2015-06-29:
Hi Tony,
Thanks for your reply.
I have gotten the data between brackets to send, but as you confirmed, imei and time/date isn't passed on.
It would be great if you could give us some hints to get these sent.

Cheers!

TonyArmitstead, 2015-06-29:
I have created a fork of this project and uploaded all my changes to it.
https://github.com/TonyArmitstead/opentracker.git
You are welcome to clone or just take a look.
NOTE: I am nowhere near finished making my changes yet - dont expect a working system from this repository - yet.

m4rkw, 2015-07-08:
FWIW the original bug of this thread is fixed in master now.  Tony - what other issues are you having?  Let me know and I'll investigate.

m4rkw, 2015-07-08:
Oh also the gps time parsing code should be fixed now too (someone else submitted a patch).  Funny I noticed ages ago that the timestamps from the gps looked a bit weird, didn't even occur to me that the code might be broken.. I just take the timestamp from the server anyway so wasn't really too bothered.

m4rkw, 2015-07-08:
Tried to read your forked commit but so many of the changes are just formatting so it's a bit hard to see the code diffs..

TonyArmitstead, 2015-07-09:
Hi Mark, I am working my way through the code re-implementing a good chunk of it to remove potential bugs (i.e. code issues I see through inspection) and to remove actual bugs. Since I found myself changing lots of code I decided to reformat it all to make it readable *to me* and to get rid of the large sections of commented out code. I know this makes diffs hard, but the code was, *IMO*, very poorly formatted. I will keep pushing my changes into my fork until I am happy that the code basically works - which it still does not.

I am also beginning to suspect problems in the platform integration. I regularly see missing data from the modem which leads me to suspect the uart code is broken. I also see receive data corrupted by transmitted data - especially in the TCP transactions. Both of these show themselves in modem time out messages.

Further, my unit stops sending after a few days - so I suspect time interval issues, but I have not investigated this yet.

Thanks for your interest. Tony

m4rkw, 2015-07-09:
Hi Tony,

It's almost certainly related to changes I made recently in an attempt to optimise the codebase.  I agree it's poorly formatted, I tried to keep to the (bad) formatting when I pushed commits through to make it easier to see the diffs but I completely agree it's a nightmare to work with.  I've been using the new raw sending mode that I added without issue for many weeks, so it seems like the bugs are specific to the http-posting mode.  I'm working through some bugs with someone else in another issue thread on here, once I get through those it will be working in both modes.

What do you mean by 'data missing from the modem'?  I suspect that the other issue where you see received data corrupted by transmitted data is related to this commit:

https://github.com/geolink/opentracker/commit/757fbae5f28b4551c9bbb4f00370fcb03ea586d5

Before that, there were delay(n) statements everywhere in order to wait for the final response from the modem.  I stripped most of these out and wrote some code to detect when the modem had finished responding, but it seems it's buggy.  The last couple of commits I pushed yesterday and today are related to this - some functions where the code is expecting to get a multi-line buffer rather than just the final line from the modem are broken.  You may find the code works better if you check out the version just before this commit, but in any case I'm going to keep on working to fix it.

Mark

Fyod, 2015-07-09:
I'm really happy I'm able to get data sent to my SQL server, so a BIG thank you to you guys taking your time to make this code.
If you need any ideas for the calls and SMS messaging, have a look at www.gsmlib.org. The code is written for SIM908, but most of the AT commands are the same for Quectel calls/messaging. That part of the gsmlib code worked flawlessly for me, but HTTP was a problem.

Cheers!

m4rkw, 2015-07-09:
Glad you've got it working Fyod :)

Fyod, 2015-07-15:
Would anyone happen to know how to add a string after GPS data?
For example: [...gps data...,1.0,2.0,3.0,4.0]
The string alteady has commas and looks like this: 1.0,2.0,3.0,4.0 so I just need to add it in there.

m4rkw, 2015-07-15:
@Fyod see: https://github.com/geolink/opentracker/blob/master/Opentracker_3_0_1/data.ino#L53-L70

m4rkw, 2015-07-23:
HTTP mode is reportedly working now, going to close this issue for now.

closed 2015-07-23 by m4rkw