geolink/opentracker #14: Bug in new gsm code. Buffer is overwriting.

ilevans, 2015-07-07:
The new code has a bug in it that causes gsm communication not work. The modem_reply buffer is being overwritten causing required data to be lost.

Using the code to retreive the IMEI number . . .

gsm_get_imei() 
calls
gsm_wait_for_reply(1)
which calls
get_gsm_reply() multiple times
which stored data in modem_reply buffer from the befinning.

When control arrives back at gsm_get_imei() the required data is no longer in the buffer. 

DEBUG OUTPUT BELOW
========================================
gsm_get_imei() started
Modem Reply:
AT+GSN

Modem Reply:
863071016800770

Modem Reply:

Modem Reply:
OK

gsm_get_imei() result:
.
gsm_get_imei() completed
========================================

m4rkw, 2015-07-08:
now fixed in master, will look at your other issue shortly

Mark

ilevans, 2015-07-08:
Still not picking up the IMEI as it is picking up the incorrect modem response.

Debug Output Below
==========
gsm_get_imei() started
Modem Reply:
AT+GSN

gsm_get_imei() result:
O
gsm_get_imei() completed
gsm_startup_cmd() started
Modem Reply:
863071016800770

Modem Reply:

Modem Reply:
OK

Modem Reply:
AT+QISDE=0

Modem Reply:
OK

Modem Reply:

Modem Reply:
+CFUN: 1

Modem Reply:
AT+QINDI=1

Modem Reply:
OK

==========

m4rkw, 2015-07-08:
Argh sorry, try now?

Mark

ilevans, 2015-07-08:
Getting there.

Now the data being sent is strange.

Debug Output Below
====
collect_gps_data(): fix acquired
Failed checksums:
0
collect_all_data() completed
Current:
�.[,80715,12545209,-26.619892,152.859207,5.67,442.50,162.52,161,5]

====

There also appears to be an issue with retrieving the time from the gsm modem. 

Debug Output Below
====
gsm_get_time() started
Modem Reply:
AT+QI
Modem Reply:
DEAC
Modem Reply:
T
AT+CCLK?

Modem Reply:
+CCLK: "08/07/15,12:55:07+00"

Modem Reply:

Modem Reply:
OK

gsm_get_time() result:
�.0
gsm_get_time() completed

m4rkw, 2015-07-08:
Hi,

I've reverted the earlier patch made to fix the gps date/time.  Please could you try again and post the output?  I'm pretty sure it'll still be broken but I want to rule out additional impact from the earlier patch.

Mark

ilevans, 2015-07-08:
Hi Mark

Sorry for the delay, sleep beckoned and I obliged.

I have included the full debug log this time to assist your efforts as requested. 

As you expected, still broken.

Ian 
(GMT +10)

Full Debug Output Below
----
gsm_on_off() finished
gsm_on_off() started
gsm_on_off() finished
gsm_restart() completed
gsm_send_at() started
gsm_send_at() completed
gsm_send_at() started
gsm_send_at() completed
gsm_set_pin() started
gsm_set_pin(): PIN is not requered
gsm_set_pin() completed
gsm_get_imei() started
Modem Reply:
AT+GSN

863071016800770

OK

gsm_get_imei() result:
863071016800770
gsm_get_imei() completed
gsm_startup_cmd() started
Modem Reply:
AT+QISDE=0

Modem Reply:
OK

Modem Reply:

Modem Reply:
+CFUN: 1

Modem Reply:
AT+QINDI=1

Modem Reply:
OK

Modem Reply:
AT+CMGF=1

Modem Reply:
OK

gsm_startup_cmd() completed
gsm_set_apn() started
Modem Reply:
AT+QIREGAPP="mdata.net.au","",""

Modem Reply:
OK

Modem Reply:

Modem Reply:
[... 1173 lines cut ...]
OK

Modem Reply:
AT+QIRD=0,1,0,100

Modem Reply:
OK

Modem Reply:
AT+QIRD=0,1,0,100

Modem Reply:
+QIRD: 78.46.45.61:80,TCP,100

Modem Reply:
HTTP/1.1 200 OK

Modem Reply:
Server: nginx/1.8.0

Modem Reply:
Date: Wed, 08 Jul 2015 19:19:31 GMT

Modem Reply:
C

m4rkw, 2015-07-09:
Thanks, it's looking quite broken there, seems like nulls in the start of the data packet.  Just pushed a fix that should correct the gsm time collection, please could you try that and re-post the output?  I think this is going to be a fairly slow debug cycle but we'll get there :)

Mark

ilevans, 2015-07-09:
I re-loaded my working version of the software and captured a working log for reference. I have included the Broken and Working log file.

The data uploading is incorrect.

**Working from Pre Remove Delay Version with my date/time parsing.**
15/07/09,07:22:39+0[90715,7224800,-26.619829,152.859177,0.37,412.10,206.77,80,12]

**Not working from remove Delay Version**
09/07/15,07:25:44+0[,90715,7254900,-26.619944,152.858963,0.30,406.30,0.00,93,10]
* Date is the wrong format
* There is a comma after the start of the square brackets.
               
Ian

----

Here is the Broken log file.

----
gsm_on_off() finished
gsm_restart() completed
gsm_send_at() started
gsm_send_at() completed
gsm_send_at() started
gsm_send_at() completed
gsm_set_pin() started
gsm_set_pin(): PIN is not requered
gsm_set_pin() completed
gsm_get_imei() started
Modem Reply:
AT+GSN

863071016800770

OK

gsm_get_imei() result:
863071016800770
gsm_get_imei() completed
gsm_startup_cmd() started
Modem Reply:
AT+QISDE=0

Modem Reply:
OK

Modem Reply:

Modem Reply:
+CFUN: 1

Modem Reply:
AT+QINDI=1

Modem Reply:
OK

Modem Reply:
AT+CMGF=1

Modem Reply:
OK

gsm_startup_cmd() completed
gsm_set_apn() started
Modem Reply:
AT+QIREGAPP="mdata.net.au","",""

Modem Reply:
OK
[... 1092 lines cut ...]
Modem Reply:

SEND OK

gsm_validate_tcp() data delivered.
gsm_validate_tcp() completed.
gsm_send_http(): Sending IMEI and Key
Modem Reply:
AT+QISEND=39

> 
Modem Reply:

SEND OK

gsm_send_http(): Sending body
gsm_send_http(): Body packet size:
82
gsm_send_http(): Sending data chunk:
0
gsm_send_http(): chunk length:
82
gsm_send_http(): data sent.

----

m4rkw, 2015-07-09:
Thanks Ian, it's looking a bit better than last time.  Have just pushed another change which hopefully will correct the date/time (using your previous patch on the other issue thread) and a fix for the comma-separation issue.  Let me know how it goes.

Mark

ilevans, 2015-07-09:
Hi Mark

I am concentrating on what is being sent through at the moment and it is still wrong.

**Wrong** From latest Build
09/07/15,08:05:33+0[,907158053800-26.619926152.8592530.37435.000.001127]
~~~~~~~ Wrong Date Format
..................................~~~~~~~~~~~~~~~~~~~~~~~~~~~~ Wrong Commas Removed

**Correct** from Pre Delay removal
15/07/09,07:22:39+0[90715,7224800,-26.619829,152.859177,0.37,412.10,206.77,80,12]

cheers

Ian

m4rkw, 2015-07-09:
try now?

ilevans, 2015-07-09:
:+1:  It is working and updating the website. Strange that it is working with the dates at the front in different formats? I expect it is taking the date and time from within the [ ] brackets

**Current Send**
09/07/15,08:30:36+0[90715,8304100,-26.619947,152.859192,0.39,435.10,0.00,100,9]
+++++++
**Pre delay removal Send**
15/07/09,07:22:39+0[90715,7224800,-26.619829,152.859177,0.37,412.10,206.77,80,12]
+++++++

The only thing that seems out of wack is this part of the log. It is causing a long pause each time. What is it expecting?

----
Modem Reply:
HTTP/1.1 200 OK

Modem Reply:
Server: nginx/1.8.0

Modem Reply:
Date: Thu, 09 Jul 2015 
Warning: timed out waiting for last modem reply

----

m4rkw, 2015-07-09:
Hurrah :)

That looks like it's not correctly parsing the response from the server.  Could you try a raw post using curl and paste the output?

curl -i -d "imei=IMEI&key=KEY&d=DATAPACKET" -A "OpenTracker3.0" http://updates.geolink.io/index.php

replace IMEI, KEY and DATAPACKET with your imei, key and the full data packet that the opentracker is sending.

Thanks,
Mark

m4rkw, 2015-07-09:
Also, their server is almost certainly just using it's local clock as the timestamp - that's what I do on mine.  No reason to use the timestamp being posted really, unless you need to debug a very slow connection or something.

Neville77, 2015-07-09:
I have found a issue with using a server timestamp in the past that when a board has stored polls and then sends them once it has connection again the polls either all come in at the one time or incorrect time. This has caused some issues with accurate tracking.

m4rkw, 2015-07-09:
Interesting, I've never actually had that happen.  Whenever it failed to log it never seemed to successfully store anything, it just lost the data and posted a fresh one on the next cycle.  Good point though.

ilevans, 2015-07-09:
Mark

As per your request.

**Command**
curl -i -d "imei=863071016800770&key=cSQ88qShwC3V&d=09/07/15,08:33:09+0[90715,
8331400,-26.619892,152.859222,0.22,430.90,0.00,90,14]" -A "OpenTracker3.0" http://updates.geolink.io/index.php

**Output**
HTTP/1.1 200 OK
Server: nginx/1.8.0
Date: Thu, 09 Jul 2015 09:27:20 GMT
Content-Type: text/html
Transfer-Encoding: chunked
Connection: keep-alive

wrong key

m4rkw, 2015-07-09:
The server is saying the key is wrong, are you sure that's the right key for your imei in your geolink account?  You should probably change it now in any case and omit the new one in future posts ;)

ilevans, 2015-07-09:
Hi Mark

Yes, I saw that, which is why I sent the exact command line with it. It will be changed before I use the device again.

I took the key directly from my tracker.h file and have double checked it with the web site.

[image]

I am not familiar with curl so had to google it. I was hoping you would have a suggestion as to what I did wrong in the command. 

Mark. Feel free to run the curl command with corrections to complete the test. Let me know when you are done and I will change the key.

Ian

m4rkw, 2015-07-09:
I can't see anything wrong with that, no idea why it isn't working..  you'll probably have to contact Geolink and ask them why it's not working.  I'm in the geolink github organisation as of yesterday but I don't work for them or have access to any of their stuff so I can't look into that on the server side.  Sorry :(

opensourcegps, 2015-07-09:
key is wrong in curl command:
key=cSQ88qShwC3V

should be:
key=cSQ88qShwC3

m4rkw, 2015-07-09:
That dude has better eyes than me!

m4rkw, 2015-07-09:
@ilevans it's probably wrong in your tracker.h too :)

ilevans, 2015-07-09:
You are correct. 
It includes a trailing "V" in the tracker.h file also and it works!
Go figure.

opensourcegps, 2015-07-09:
Just to highlight few server side points:

1. timestamp is used from GPS data. 
2. Extra zeros are added to time/date server side:

```php
   if((strlen($gpspos_time)) == 7)
         $gpspos_time = '0'.$gpspos_time; 

   if((strlen($gpspos_date)) == 5)
         $gpspos_date = '0'.$gpspos_date; 
```
                                    	
3. It is important that reply from server is processed correctly and not split, because modem will set it's time from response and in future it will execute commands (i.e. turn off engine). I think parse_receive_reply() should be switched back to old way of processing modem replies. It looks like broken at the moment.

m4rkw, 2015-07-09:
@opensourcegps I think parse_receive_reply is expecting to have the whole reply in the modem_reply after line 26:

gsm_wait_for_reply(1,0);

so we can just change that to:

gsm_wait_for_reply(1,1);

and it will then have the whole buffer (rather than breaking at a newline), and probably work, although I'm not sure exactly what the problem you're seeing is..

opensourcegps, 2015-07-09:
@m4rkw sounds good. Let's change it so it receives full reply. The problem is as follows:

1. it does not recognize that data was transmitted. This is important because we are developing new feature that will save data to internal memory for later sending, in case this confirmation not received. This is done by parsing "#eof" from server.

2. It does not process settime command. 

Log exmples:
Working log:

```
Connection: close

#t:15/07/09,07:22:38+00
#eof
OK

0, CLOSED

0, CLOS
Found header packet
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

No more data available.
Data was fully received by the server.
parse_cmd() started
Received commands:
#t:15/07/09,07:22:38+00
#eof
Found settime command.
15/07/09,07:22:38+00
Valid time string found.
gsm_set_time() started
Modem Reply:
AT+CCLK="15/07/09,07:22:38+00"

OK

gsm_set_time() completed
parse_cmd() completed

```

Not working log:

```

parse_receive_reply() started
Modem Reply:
AT+QIRD=0,1,0,100

Modem Reply:
OK

Modem Reply:
AT+QIRD=0,1,0,100

Modem Reply:
OK

Modem Reply:
AT+QIRD=0,1,0,100

Modem Reply:
OK

Modem Reply:
AT+QIRD=0,1,0,100

Modem Reply:
OK

Modem Reply:

Modem Reply:
+QIRDI: 0,1,0

Modem Reply:
AT+QIRD=0,1,0,100

Modem Reply:
+QIRD: 78.46.45.61:80,TCP,100

Modem Reply:
HTTP/1.1 200 OK

Modem Reply:
Server: nginx/1.8.0

Modem Reply:
Date: Wed, 08 Jul 2015 
Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

Modem Reply:
+QIRD: 78.46.45.61:80,TCP,49

Modem Reply:
Connection: close

Modem Reply:

Modem Reply:
#t:15/07/08,19:18:29+00

Modem Reply:
#eof

Modem Reply:
OK

Modem Reply:

Modem Reply:
0, CLOSED

Modem Reply:

Modem Reply:
0, CLOSAT+QIRD=0,1,0,100

Modem Reply:
ERROR

No more data available.
Data was not received by the server.
parse_cmd() started
Received commands:
�7�
parse_cmd() completed

```

I hope this makes sense.

m4rkw, 2015-07-09:
Ok, pushed that, need to get back to my actual job for a bit now ;)  Mark

m4rkw, 2015-07-09:
@ilevans are you happy that this issue is resolved now?

ilevans, 2015-07-09:
:-1:  Unfortunately not @m4rkw .

It is getting stuck in a loop of timeouts.

Ian

----

Complete Log Below
----

gsm_on_off() finished
gsm_on_off() started
gsm_on_off() finished
gsm_restart() completed
gsm_send_at() started
gsm_send_at() completed
gsm_send_at() started
gsm_send_at() completed
gsm_set_pin() started
gsm_set_pin(): PIN is not requered
gsm_set_pin() completed
gsm_get_imei() started
Modem Reply:
AT+GSN

863071016800770

OK

gsm_get_imei() result:
863071016800770
gsm_get_imei() completed
gsm_startup_cmd() started
Modem Reply:
AT+QISDE=0

Modem Reply:
OK

Modem Reply:

Modem Reply:
+CFUN: 1

Modem Reply:
AT+QINDI=1

Modem Reply:
OK

Modem Reply:
AT+CMGF=1

Modem Reply:
OK

gsm_startup_cmd() completed
gsm_set_apn() started
Modem Reply:
AT+QIREGAPP="mdata.net.au","",""

Modem Reply:
OK

Modem Reply:

Modem Reply:
+CPIN: READY

[... 363 lines cut ...]
ERROR

Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

m4rkw, 2015-07-10:
Hi Ian, haven't forgotten about you just crazy busy with work atm, will try to get to this over the weekend

Mark

m4rkw, 2015-07-18:
@ilevans hey i've just pushed a change which should fix this for you, please let me know if it works.

Mark

ilevans, 2015-07-18:
@m4rkw I have downloaded and performed my first test of the revised code and still no joy.

It still appears to be looping continuously after sending to the server. (Extract Below).

----
Connected to remote server: 
updates.geolink.io
gsm_connect() completed
gsm_send_http(): sending data.
04/01/01,00:10:43+0[180715,19483610,-26.620350,152.859207,1.39,425.90,352.92,105,6]

gsm_send_http(): Length of data packet:
122
gsm_send_http(): Length of header packet:
171
Modem Reply:
AT+QISEND=171

Modem Reply:
> 
gsm_validate_tcp() started.
Modem Reply:

Modem Reply:
SEND OK

gsm_validate_tcp() data delivered.
gsm_validate_tcp() completed.
gsm_send_http(): Sending IMEI and Key
Modem Reply:
AT+QISEND=38

Modem Reply:
> 
Modem Reply:

Modem Reply:
SEND OK

gsm_send_http(): Sending body
gsm_send_http(): Body packet size:
84
gsm_send_http(): Sending data chunk:
0
gsm_send_http(): chunk length:
84
Modem Reply:
AT+QISEND=84

Modem Reply:
> 
gsm_send_http(): data sent.
Modem Reply:

SEND OK

Modem Reply:

+QIRDI: 0,1,0

Warning: timed out waiting for last modem reply
parse_receive_reply() started
Modem Reply:
AT+QIRD=0,1,0,100

+QIRD: 78.46.45.61:80,TCP,100
HTTP/1.1 200 OK
Server: nginx/1.8.0
Date: Sat, 18 Jul 2015 19:48:40 GMT
C
Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

+QIRD: 78.46.45.61:80,TCP,49
Connection: close

#t:15/07/18,19:48:40+00
#eof
OK

0, CLOSED

0, CLOS
Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

----

m4rkw, 2015-07-19:
@ilevans last night I tried switching mine into http mode and it does the same thing.  I spent some time trying to debug it but I got tripped up by a problem with my server, will hopefully have some more time during the week.  Mark

ilevans, 2015-07-19:
No worries m4rkw. I will test it when it is ready. :smile:

m4rkw, 2015-07-20:
@ilevans i've just pushed a fix blind which might help, i've put the delay statements back into the http sending function that were removed in favour of final response detection.  When you have time could you give it a try and post the output?  I'm away from home at the moment so can't test it myself.

ilevans, 2015-07-20:
Compile Error
----
gsm.ino: In function 'void gsm_send_http_current()':
gsm:539: error: too few arguments to function 'void gsm_get_reply(int)'
Opentracker_3_0_1.ino:42:6: note: declared here
Exception in thread "Thread-22" java.lang.NullPointerException

m4rkw, 2015-07-21:
Argh sorry stupid type :S  try now?

m4rkw, 2015-07-21:
typo haha. oh dear.  i think i need coffee

ilevans, 2015-07-21:
Still Looping as before

Log file below
----
gsm_on_off() finished
gsm_restart() completed
gsm_send_at() started
gsm_send_at() completed
gsm_send_at() started
gsm_send_at() completed
gsm_set_pin() started
gsm_set_pin(): PIN is not requered
gsm_set_pin() completed
gsm_get_imei() started
Modem Reply:
AT+GSN

863071016800770

OK

gsm_get_imei() result:
863071016800770
gsm_get_imei() completed
gsm_startup_cmd() started
Modem Reply:
AT+QISDE=0

Modem Reply:
OK

Modem Reply:

Modem Reply:
+CFUN: 1

Modem Reply:
AT+QINDI=1

Modem Reply:
OK

Modem Reply:
AT+CMGF=1

Modem Reply:
OK

gsm_startup_cmd() completed
gsm_set_apn() started
Modem Reply:
AT+QIREGAPP="mdata.net.au","",""

Modem Reply:
OK

Modem Reply:

Modem Reply:
+CPIN: READY

Modem Reply:
AT+QIDNSCFG="8.8.8.8"

Modem Reply:
OK

Modem Reply:
AT+QIDNSIP=1

[... 281 lines cut ...]
ERROR

Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

m4rkw, 2015-07-22:
Ok try again?

ilevans, 2015-07-22:
Compile error. . . Need a "}" after gsm:693 by my reckoning.

Given that change, same problem.

Log follows
----

gsm_startup_cmd() started
gsm_on_off() finished
gsm_on_off() started
gsm_on_off() finished
gsm_restart() completed
gsm_send_at() started
gsm_send_at() completed
gsm_send_at() started
gsm_send_at() completed
gsm_set_pin() started
gsm_set_pin(): PIN is not requered
gsm_set_pin() completed
gsm_get_imei() started
Modem Reply:
AT+GSN

863071016800770

OK

gsm_get_imei() result:
863071016800770
gsm_get_imei() completed
gsm_startup_cmd() started
Modem Reply:
AT+QISDE=0

Modem Reply:
OK

Modem Reply:

Modem Reply:
+CFUN: 1

Modem Reply:
AT+QINDI=1

Modem Reply:
OK

Modem Reply:
AT+CMGF=1

Modem Reply:
OK

gsm_startup_cmd() completed
gsm_set_apn() started
Modem Reply:
AT+QIREGAPP="mdata.net.au","",""

Modem Reply:
OK

Modem Reply:

Modem Reply:
+CPIN: READY

Modem Reply:
AT+QIDNSCFG="8.8.8.8"

[... 252 lines cut ...]
ERROR

Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

m4rkw, 2015-07-23:
Ok one more time?  I've put the bit of code that parses the http reply back to how it was before all the delay statements were taken out, so it should at least do something different this time..

ilevans, 2015-07-23:
m3rkw :thumbsup:

Things are looking good for the sending of the data.
I will carry out further testing as time permits.

I now consider this issue resolved. :smile:

thanks

@ilevans 

Log output follows (2 sends)
----
gsm_on_off() finished
gsm_restart() completed
gsm_send_at() started
gsm_send_at() completed
gsm_send_at() started
gsm_send_at() completed
gsm_set_pin() started
gsm_set_pin(): PIN is not requered
gsm_set_pin() completed
gsm_get_imei() started
Modem Reply:
AT+GSN

863071016800770

OK

gsm_get_imei() result:
863071016800770
gsm_get_imei() completed
gsm_startup_cmd() started
Modem Reply:
AT+QISDE=0

Modem Reply:
OK

Modem Reply:

Modem Reply:
+CFUN: 1

Modem Reply:
AT+QINDI=1

Modem Reply:
OK

Modem Reply:
AT+CMGF=1

Modem Reply:
OK

gsm_startup_cmd() completed
gsm_set_apn() started
Modem Reply:
AT+QIREGAPP="mdata.net.au","",""

Modem Reply:
OK

Modem Reply:

Modem Reply:
+CPIN: READY

[... 623 lines cut ...]
$GNRMC,074749.089,A,2637.1944,S,15251.5519,E,0.29,0.00,230715,,,A*6E
Invalid altitude, retrying.

$GPVTG,0.00,T,,M,0.29,N,0.53,K,A*30
$GPGGA,074749.089,2637.1944,S,15251.5519,E,1,7,1.90,430.6,M,39.7,M,,*48
Invalid course, retrying.

$GNGSA,A,3,07,16,09,30,,,,,,,,,2.11,1.90,0.91*14
$GNGSA,A,3,68,78,69,,,,,,,,,,2.11,1.90,0.91*10
$GPGSV,3,1,09,09,71,180,20,42,55,333,28,07,47,223,46,19,29,068,25*76
$GPGSV,3,2,09,30,26,257,31,16,20,142,28,11,02,028,,193,,,*7F
$GPGSV,3,3,09,23,,,20*73
$GLGSV,2,1,05,68,71,269,40,78,44,185,26,69,26,214,29,88,15,079,*68
$GLGSV,2,2,05,87,09,126,24*55
$GNGLL,2637.1944,S,15251.5519,E,074749.089,A,A*50
$GNRMC,074751.000,A,2637.1947,S,15251.5508,E,0.42,0.00,230715,,,A*68
Invalid course, retrying.

$GPVTG,0.00,T,,M,0.42,N,0.78,K,A*34
$GPGGA,074751.000,2637.1947,S,15251.5508,E,1,7,1.90,431.6,M,39.7,M,,*42
Invalid course, retrying.

$GNGSA,A,3,07,16,09,30,,,,,,,,,2.11,1.90,0.91*14
$GNGSA,A,3,68,78,69,,,,,,,,,,2.11,1.90,0.91*10
$GPGSV,3,1,09,09,71,180

m4rkw, 2015-07-23:
Awesome news :)  Sorry it took so long to resolve, I've been crazy with work these last couple of weeks getting ready for a launch and I'm off on holiday to France next week.  Really glad I got this fixed for you before I go!

Mark

closed 2015-07-23 by m4rkw