geolink/opentracker #15: Parsing gps date and time.

ilevans, 2015-07-07:
The parsing of the gps time data does not work for 00:00 to 00:59 UTC time.

I am in Australia and 00:00 UTC is 10:00 am. Data from 10:00 to 11:00 is corruptly translated to being in the future:

I have replaced the gps date time parsing with the following. 

Sorry, I do not have GIT installed as yet.

gps.ino from line 128
============================================================

             //update current time var - format 04/12/98,00:35:45+00
             // Add 1000000 to ensure the position of the digits
             ltoa(date_gps+1000000, tmp, 10);  //1ddmmyy                    
             time_char[0] = tmp[1]; 
             time_char[1] = tmp[2]; 
             time_char[2] = '/'; 
             time_char[3] = tmp[3]; 
             time_char[4] = tmp[4]; 
             time_char[5] = '/'; 
             time_char[6] = tmp[5]; 
             time_char[7] = tmp[6]; 
             time_char[8] = ',';                    
          
             
             // Add 1000000 to ensure the position of the digits
             ltoa(time_gps + 100000000, tmp, 10);  //1hhmmssms
             time_char[9] = tmp[1]; 
             time_char[10] = tmp[2]; 
             time_char[11] = ':';              
             time_char[12] = tmp[3]; 
             time_char[13] = tmp[4]; 
             time_char[14] = ':';              
             time_char[15] = tmp[5]; 
             time_char[16] = tmp[6]; 
             time_char[17] = '+';              
             time_char[18] = '0';
             time_char[19] = '0';
             time_char[20] = '\0';

============================================================

m4rkw, 2015-07-08:
Aha, i wondered why the timestamps from the GPS looked odd :D

m4rkw, 2015-07-08:
Nice one, have sent a pull request for you

closed 2015-07-08 by m4rkw