geolink/opentracker #20: Failing to retreive IMEI

ilevans, 2015-07-18:
Hi All

On startup after downloading the OpenTacker 301 software the startup re-acts differently to a cold start. (refer to Bug in new gsm code. Buffer is overwriting)

Retrieval of the IMEI fails.
Many timeouts occur on the modem and even stranger things occur.

Extract of log is below
----
gsm_on_off() finished
gsm_on_off() started
gsm_on_off() finished
gsm_restart() completed
gsm_send_at() started
gsm_send_at() completed
gsm_send_at() started
gsm_send_at() completed
gsm_set_pin() started
gsm_set_pin(): PIN is not requered
gsm_set_pin() completed
gsm_get_imei() started
gsm_get_imei() result:
/
gsm_get_imei() completed
gsm_startup_cmd() started
Warning: timed out waiting for last modem reply
Warning: timed out waiting for last modem reply
Warning: timed out waiting for last modem reply
gsm_startup_cmd() completed
gsm_set_apn() started
Warning: timed out waiting for last modem reply
Warning: timed out waiting for last modem reply
Warning: timed out waiting for last modem reply
gsm_set_apn() completed
store_get_index() started
store_get_index(): Found log position:
131320
store_get_index() ended
setup() completed
Ignition status:
0
Ignition is ON!
collect_all_data() started
gsm_get_time() started
Warning: timed out waiting for last modem reply
gsm_get_time() result:
/
gsm_get_time() completed

Fyod, 2015-07-21:
Try this:

void gsm_get_imei() {
  for (int a = 0; a < 3; a++) {
    int i;
    char preimei[20];             //IMEI number
    debug_println(F("gsm_get_imei() started"));
    gsm_get_reply(0);
    //get modem's imei 
    snprintf(modem_command, sizeof(modem_command),
        "AT+GSN");
    gsm_send_command();
    gsm_wait_for_reply(1,0);
    delay(1000);
    //reply data stored to modem_reply[200]
    char *tmp = strstr(modem_reply, "AT+GSN\r\r\n");
    tmp += strlen("AT+GSN\r\r\n");
    char *tmpval = strtok(tmp, "\r");
    //copy data to main IMEI var
    for (i = 0; i < strlen(tmpval); i++) {
        preimei[i] = tmpval[i];
        if (i > 17) { //imei can not exceed 20 chars
            break;
        }
    }
    //null terminate imei
    preimei[i + 1] = '\0';
    debug_println(F("gsm_get_imei() result:"));
    debug_println(preimei);
    memcpy(config.imei, preimei, 20);
    if (strlen(preimei) < 15) {
      debug_println(F("Invalid IMEI, retrying."));
      a = 0; //reset main try counter
      continue;
    }
    debug_println(F("gsm_get_imei() completed"));
  }
}

ilevans, 2015-07-23:
This no longer appears to be an issue after testing the buffer overwrite changes.

ilevans, 2015-07-23:
Closing Issue.

closed 2015-07-23 by ilevans