geolink/opentracker #21: CMS errors?

Fyod, 2015-07-21:
Anyone else experiencing CMS errors, usually 604 and 10 on SMS?
Not sure if this is necessarily coding issue or network/module.
I have seen some incoming SMS printed in Serial Monitor a couple times.

m4rkw, 2015-07-21:
I get one on startup, I think it's due to not correctly verifying the previous command.  SMS works perfectly after startup though.  Are you using raw mode or http?  I think http is currently broken, if you're using raw mode do sms commands work?

Fyod, 2015-07-21:
Using http. I'll test raw. Thanks!

m4rkw, 2015-07-23:
Hi Fyod, http mode should be working now.  If you're still experiencing this issue please could you let me know if it just happens on the startup sms check, or whether it happens continuously, and in either case post the output of the serial console (ideally with all the logs from startup of the tracker) so I can look into it for you?

Thanks,
Mark

Fyod, 2015-07-23:
It looks like the buffer isn't cleared before sms check.

sms_check() started
th Suhosin-Patch
X-Powered-By: PHP/5.2.6-1+lenny9
AccAT+CMGL="REC UNREAD"

+CMS ERROR: 604
Deleting READ and SENT SMS
Modem Reply:
AT+QMGDA="DEL READ"

Modem Reply:
+CMS ERROR: 604

Modem Reply:
AT+QMGDA="DEL SENT"

Modem Reply:
+CMS ERROR: 604

sms_check() completed

Fyod, 2015-07-24:
Looks like that may have been my mistake, but now I get:

```
sms_check() started
AT+CMGL="REC UNREAD"

+CMS ERROR: 3

Call Ready
Deleting READ and SENT SMS
Modem Reply:
AT+QMGDA="DEL READ"

Modem Reply:
+CMS ERROR: 604

Modem Reply:
AT+QMGDA="DEL SENT"

Modem Reply:
+CMS ERROR: 604
```

Fyod, 2015-07-25:
Issue resolved.
I had updated my code with Tony's update (https://github.com/TonyArmitstead/opentracker/commit/864ddd0126350bc58ece3758565b72a246f8a0ba#diff-23768ff6988f7dc3e845e57cb61504e0R126) 
and I missed two lines, surprisingly didn't get an error.

Btw: where can I submit code contribution? I made some code to send network name and battery status (percentage, voltage) in the data packet when running off a lipo.

Fyod, 2015-07-25:
So now that that works, I found that enabling GSM_STAY_ONLINE 1 causes the data to be sent once, then it fails because the modem returns ALREADY CONNECT
```
collect_all_data() completed
Current: 
*** datat here is correct ***

gsm_send_at() started
gsm_send_at() completed
Modem Reply:
AT
AT

Modem Reply:
OK

gsm_disconnect() Disabled
GPRS deactivated.
gsm_connect() started
Connecting to remote server...
0
Modem Reply:

Modem Reply:
OK

Modem Reply:
AT+QIOPEN="TCP","foo.foo.com","80"

Modem Reply:
ALREADY CONNECT

Modem Reply:

Modem Reply:
ERROR

Can not connect to remote server: 
foo.foo.com
Connecting to remote server...
1
Modem Reply:
AT+QIOPEN="TCP","foo.foo.com","80"

Modem Reply:
ALREADY CONNECT

Modem Reply:

Modem Reply:
ERROR

Can not connect to remote server: 
146.255.26.38
gsm_connect() completed
Error, can not send data, no connection.
Can not send data
Sleeping for: 3647ms
```

ilevans, 2015-07-26:
Hi @Fyod 

The updates.geolink.io site closes the connection so that another can be opened. 

If the site you are using does not close the connection, then it would be necessary to issue a AT+QICLOSE instead of the AT+QIDEACT.

Extract from one of my test logs
--
gsm_send_at() completed
Modem Reply:
AT
AT

Modem Reply:
OK

gsm_disconnect() Disabled
GPRS deactivated.
gsm_connect() started
Connecting to remote server...
0
Modem Reply:

Modem Reply:
OK

Modem Reply:
AT+QIOPEN="TCP","updates.geolink.io","80"

Modem Reply:
OK

Modem Reply:

Modem Reply:
CONNECT OK

Connected to remote server: 
updates.geolink.io
gsm_connect() completed
gsm_send_http(): sending data.
15/07/26,00:29:09+0[260715,292000,-26.619543,152.859207,0.80,448.00,0.00,142,5]

gsm_send_http(): Length of data packet:
118
gsm_send_http(): Length of header packet:
171
Modem Reply:
AT+QISEND=171

Modem Reply:
> 
gsm_validate_tcp() started.
Modem Reply:

Modem Reply:
SEND OK

gsm_validate_tcp() data delivered.
gsm_validate_tcp() completed.
gsm_send_http(): Sending IMEI and Key
Modem Reply:
AT+QISEND=38

Modem Reply:
> 
Modem Reply:

Modem Reply:
SEND OK

gsm_send_http(): Sending body
gsm_send_http(): Body packet size:
80
gsm_send_http(): Sending data chunk:
0
gsm_send_http(): chunk length:
80
gsm_send_http(): data sent.
parse_receive_reply() started
Modem Reply:
AT+QISEND=80

> 
Modem Reply:

SEND OK

Modem Reply:
AT+QIRD=0,1,0,100

OK

+QIRDI: 0,1,0

Modem Reply:
AT+QIRD=0,1,0,100

+QIRD: 78.46.45.61:80,TCP,100
HTTP/1.1 200 OK
Server: nginx/1.8.0
Date: Sun, 26 Jul 2015 00:29:24 GMT
C
Modem Reply:
AT+QIRD=0,1,0,100

+QIRD: 78.46.45.61:80,TCP,49
Connection: close

#t:15/07/26,00:29:24+00
#eof
OK

0, CLOSED

0, CLOS
Found header packet
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

No more data available.
Data was fully received by the server.

--

Fyod, 2015-07-26:
Wonderful, going to try that today! Thank you!

Fyod, 2015-07-28:
Same problem with QICLOSE. 
It looks as if the code doesn't accept the Already connected reply.

ilevans, 2015-07-28:
I have put links to the GSM and GPS manuals on the wiki. 
It sounds like the close is not working. 
Alternatively you could look at changing the cost to recognise the Already connected reply as well.

opensourcegps, 2015-07-29:
```
#define GSM_STAY_ONLINE 1 
```
breaks sms checks with:
+CMS ERROR: 604

Set to 0 by default at the moment and need further investigating, possibly modem can not return SMS when connected to remove server.

m4rkw, 2015-08-03:
@opensourcegps I think 604 is "cannot allocate control socket" so i think your guess is probably right.  I guess users who want to use that option could implement sms handling server-side and then feed the commands back to the tracker in the http response.  Infact, thinking about it, I think that might already just work if #<command> is returned by the server rather than #eof (or maybe preceding it).

Fyod, 2015-08-03:
I agree, SMS is working OK when there's a disconnect or if the device is restarted.
# should be returned by the server as the first character in the header?

ppescher, 2016-07-01:
Recent code issues AT+QICLOSE instead of AT+QIDEACT when GSM_STAY_ONLINE=1, so the GPRS session is maintained but TCP connection is closed anyway.
However, the server does close the TCP connection before the client does, so that error might be related to something else.
I never saw it happen.

closed 2016-07-01 by ppescher