geolink/opentracker #24: WEBSITE Errors in Interpreting GPS Date/Time

ilevans, 2015-07-26:
Even with the correction that I have logged a PULL request for, there still appears to be issues with the website interpreting the data that is provided. My testing today before the correction took place between 25-Jul-2015 23:50 and 26-Jul-2015 00:30:00 (GMT Time). The data between 00:00:00 and 00:30:00 (GMT) was misinterpreted and appear on the website up to 3 days in the future

IE: 26-Jul-2015 00:29:40 (GMT)
15/07/26,00:29:35+0[260715,294000,-26.620186,152.859207,0.80,421.50,141.92,142,5]

----
The corrected data was sent between 26-Jul-2015 00:44:00 and 26-Jul-2015 00:46:00 (GMT Time). This data I cannot locate on the website. The latest reading however shows up correctly.

IE: 26-Jul-2015 00:44:47 (GMT)
15/07/26,00:44:47+0[260715,00445400,-26.620241,152.858627,2.98,445.20,60.04,303,6]

Neville77, 2015-07-29:
I am also having trouble with the time & date with the latest code. It only seems to provide the date a couple of days in the future, and most of the polls do not succesfully get to the geolink server.

opensourcegps, 2015-07-29:
This issue has been resolved server side. If problem still exists, please re-open the issue.

closed 2015-07-29 by opensourcegps