geolink/opentracker #25: New Version not updating website.

Neville77, 2015-07-29:
I have updated to the newest version and got not get sucessfull polling to the Geolink website.

PWRMON is LOW. Modem is not running.
gsm_on_off() started
gsm_on_off() finished
gsm_restart() completed
gsm_send_at() started
gsm_send_at() completed
gsm_send_at() started
gsm_send_at() completed
gsm_set_pin() started
gsm_set_pin(): PIN is not requered
gsm_set_pin() completed
gsm_get_imei() started
Modem Reply:
AT+GSN

863071016800903

OK

+CFUN: 1

+CPIN: READY

gsm_get_imei() result:
863071016800903
gsm_get_imei() completed
gsm_startup_cmd() started
Modem Reply:
AT+QISDE=0

Modem Reply:
OK

Modem Reply:
AT+QINDI=1

Modem Reply:
OK

Modem Reply:
AT+CMGF=1

Modem Reply:
OK

gsm_startup_cmd() completed
gsm_set_apn() started
Modem Reply:
AT+QIRE{ÿ
Warning: timed out waiting for last modem reply
Modem Reply:
AT+QIDNSCFG=
Modem Reply:
"8.8.8.8"

Modem Reply:
OK

Modem Reply:
AT+QIDNSIP=1

Modem Reply:
OK

gsm_set_apn() completed
store_get_index() started
store_get_index(): Found log position:
248
[... 784 lines cut ...]
gsm_disconnect(): DEACT OK not found.
gsm_disconnect() completed
Can not send data, saving to flash memory
Ignition status:
0
Ignition is ON!
collect_all_data() started
gsm_get_time() started
Modem Reply:
AT+QIDEACT

Modem Reply:
AT+CCLK?

+CCLK: "29/07/15,04:53:57+00"

OK

gsm_get_time() result:
29/07/15,04:53:57+0
gsm_get_time() completed
$GNGSA,A,3,26,01,07,03,32,06,09,16,23,193,,,1.23,0.90,0.83*2C
$$GNRMC,045358.000,A,3800.7264,S,14515.9014,E,0.25,95.62,290715,,,A*51
GPS fix received.
Data is current.

opensourcegps, 2015-07-29:
Was it working before the update?

It looks like an issue with internet connection. Are you sure your SIM card has internet? There is some strange header in reply "X-Cache: MISS", which is not supplied by us, this is something from your provider. Can you please make sure you are using correct APN and device can get online properly.

Here is current log from commit f5a625a113eb5c9fed418ac09acbec8be736f007

```
Connected to remote server: 
updates.geolink.io
gsm_connect() completed
gsm_send_http(): sending data.
15/07/29,06:28:18+0[290715,6282300,11.182072,22.353655,0.43,215.30,247.96,255,6]

gsm_send_http(): Length of data packet:
110
gsm_send_http(): Length of header packet:
171
Modem Reply:
AT+QISEND=171

Modem Reply:
> 
gsm_validate_tcp() started.
Modem Reply:

Modem Reply:
SEND OK

gsm_validate_tcp() data delivered.
gsm_validate_tcp() completed.
gsm_send_http(): Sending IMEI and Key
Modem Reply:
AT+QISEND=29

Modem Reply:
> 
Modem Reply:

Modem Reply:
SEND OK

gsm_send_http(): Sending body
gsm_send_http(): Body packet size:
81
gsm_send_http(): Sending data chunk:
0
gsm_send_http(): chunk length:
81
gsm_send_http(): data sent.
parse_receive_reply() started
Modem Reply:
AT+QISEND=81

> 
Modem Reply:

SEND OK

+QIRDI: 0,1,0

Modem Reply:
AT+QIRD=0,1,0,100

+QIRD: 78.46.45.61:80,TCP,100
HTTP/1.1 200 OK
Server: nginx/1.8.0
Date: Wed, 29 Jul 2015 06:28:27 GMT
Content-Type: text/html

OK

Modem Reply:
AT+QIRD=0,1,0,100

+QIRD: 78.46.45.61:80,TCP,49
Connection: close

#t:15/07/29,06:28:27+00
#eof
OK

43, CLOSED

43, CLOSED

Found header packet
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

No more data available.
Data was fully received by the server.
parse_cmd() started
Received commands:
#t:15/07/29,06:28:27+00
#eof
Found settime command.
15/07/29,06:28:27+00
Valid time string found.
gsm_set_time() started
Modem Reply:
AT+CCLK="15/07/29,06:28:27+00"

Modem Reply:
OK

gsm_set_time() completed
parse_cmd() completed
parse_receive_reply() completed
gsm_disconnect() started
gsm_disconnect(): DEACT OK not found.
gsm_disconnect() completed
Data sent successfully.
```

Neville77, 2015-07-30:
Hi,
The old version does update the website but has the X-Cache: MISS as well

[image]

Sleeping for:
-2516
ms
Error: negative sleep time.
sms_check() started
Modem Reply:

AT+CMGL="REC UNREAD"

OK
Deleting READ and SENT SMS
Modem Reply:
AT+QMGDA="DEL READ"

OK

Modem Reply:
AT+QMGDA="DEL SENT"

OK

sms_check() completed
Ignition status:
0
Ignition is ON!
collect_all_data() started
gsm_get_time() started
Modem Reply:

Modem Reply:
AT+CCLK?

+CCLK: "30/07/15,05:06:51+00"

OK

gsm_get_time() result:
30/07/15,05:06:51+0
gsm_get_time() completed
$GPVTG,70.13,T,,M,0.01,N,0.02,K,D*0E
$GPGGA,050620.000,3800.7827,31,65,34,054,42*68
$GLGSV,2,2,07,67,29,215,32,75,23,148,31,83,01,204,18*56
$GNGLL,3800.7289,S,14515.9039,E,050652.000,A,D*53
$GNRMC,050653.000,A,3800.7289,S,14515.9039,E,0.01,70.13,300715,,,D*51
GPS fix received.
Data is current.
Current time set from GPS time:
30/07/15,05:06:53+00
gsm_set_time() started
Modem Reply:
AT+CCLK="30/07/15,05:06:53+00"

OK

gsm_set_time() completed

GPS already available, breaking
collect_gps_data(): fix acquired
Failed checksums:
7
collect_all_data() completed
Current:
30/07/15,05:06:51+0[300715,5065300,-38.012150,145.265060,0.02,43.30,70.13,64,17]

gsm_send_at() started
Modem Reply:
AT

OK
AT

OK

gsm_send_at() completed
gsm_disconnect() started
Modem Reply:
AT+QIDEACT

DEACT OK

gsm_disconnect(): DEACT OK found
gsm_disconnect() completed
GPRS deactivated.
gsm_connect() started
Connecting to remote server...
0
Modem Reply:
AT+QIOPEN="TCP","updates.geolink.io","80"

OK

CONNECT OK

Connected to remote server: 
updates.geolink.io
gsm_connect() completed
gsm_send_http(): sending data.
30/07/15,05:06:51+0[300715,5065300,-38.012150,145.265060,0.02,43.30,70.13,64,17]

gsm_send_http(): Length of data packet:
119
gsm_send_http(): Length of header packet:
171
Modem Reply:
AT+QISEND=171

> 
gsm_validate_tcp() started.
Modem Reply:

SEND OK

gsm_validate_tcp() data delivered.
gsm_validate_tcp() completed.
gsm_send_http(): Sending IMEI and Key
Modem Reply:
AT+QISEND=38

> 
Modem Reply:

SEND OK

gsm_send_http(): Sending body
gsm_send_http(): Body packet size:
81
gsm_send_http(): Sending data chunk:
0
gsm_send_http(): chunk length:
81
gsm_send_http(): data sent.
parse_receive_reply() started
Modem Reply:
AT+QISEND=81

> 
SEND OK

+QIRDI: 0,1,0

Modem Reply:
AT+QIRD=0,1,0,100

+QIRD: 78.46.45.61:80,TCP,100
HTTP/1.1 200
Modem Reply:
AT+QIRD=0,1,0,100

+QIRD: 78.46.45.61:80,TCP,82
X-Cache: MISS
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

No more data available.
Data was not received by the server.
parse_cmd() started
Received commands:
6
parse_cmd() completed
parse_receive_reply() completed
Can not send data, saving to flash memory
Sleeping for:
-2454
ms
Error: negative sleep time.
sms_check() started
Modem Reply:

AT+CMGL="REC UNREAD"

OK
Deleting READ and SENT SMS
Modem Reply:
AT+QMGDA="DEL READ"

OK

Modem Reply:
AT+QMGDA="DEL SENT"

OK

sms_check() completed
Ignition status:
0
Ignition is ON!
collect_all_data() started
gsm_get_time() started
Modem Reply:

Modem Reply:
AT+CCLK?

+CCLK: "30/07/15,05:07:24+00"

OK

gsm_get_time() result:
30/07/15,05:07:24+0
gsm_get_time() completed
$GPVTG,70.13,T,,M,0.01,N,0.01,K,D*0D
$GPGGA,050653.000,3800.7D$GLGSV,2,1,07,76,78,149,28,66,76,146,37,77,41,326,28,65,33,054,39*6F
$GLGSV,2,2,07,67,29,215,32,75,23,148,33,83,01,205,17*5A
$GNGLL,3800.7289,S,14515.9040,E,050725.000,A,D*5C
$GNRMC,050726.000,A,3800.7289,S,14515.9040,E,0.01,70.13,300715,,,D*5C
GPS fix received.
Data is current.
Current time set from GPS time:
30/07/15,05:07:26+00
gsm_set_time() started
Modem Reply:
AT+CCLK="30/07/15,05:07:26+00"

OK

gsm_set_time() completed

GPS already available, breaking
collect_gps_data(): fix acquired
Failed checksums:
7
collect_all_data() completed
Current:
30/07/15,05:07:24+0[300715,5072600,-38.012150,145.265060,0.02,43.30,70.13,64,17]

gsm_send_at() started
Modem Reply:
AT

OK
AT

OK

gsm_send_at() completed
gsm_disconnect() started
Modem Reply:
AT+QIDEACT

DEACT OK

gsm_disconnect(): DEACT OK found
gsm_disconnect() completed
GPRS deactivated.
gsm_connect() started
Connecting to remote server...
0
Modem Reply:
AT+QIOPEN="TCP","updates.geolink.io","80"

OK

CONNECT OK

Connected to remote server: 
updates.geolink.io
gsm_connect() completed
gsm_send_http(): sending data.
30/07/15,05:07:24+0[300715,5072600,-38.012150,145.265060,0.02,43.30,70.13,64,17]

gsm_send_http(): Length of data packet:
119
gsm_send_http(): Length of header packet:
171
Modem Reply:
AT+QISEND=171

> 
gsm_validate_tcp() started.
Modem Reply:

SEND OK

gsm_validate_tcp() data delivered.
gsm_validate_tcp() completed.
gsm_send_http(): Sending IMEI and Key
Modem Reply:
AT+QISEND=38

> 
Modem Reply:

SEND OK

gsm_send_http(): Sending body
gsm_send_http(): Body packet size:
81
gsm_send_http(): Sending data chunk:
0
gsm_send_http(): chunk length:
81
gsm_send_http(): data sent.
parse_receive_reply() started
Modem Reply:
AT+QISEND=81

> 
SEND OK

+QIRDI: 0,1,0

Modem Reply:
AT+QIRD=0,1,0,100

+QIRD: 78.46.45.61:80,TCP,100
HTTP/1.1 200
Modem Reply:
AT+QIRD=0,1,0,100

+QIRD: 78.46.45.61:80,TCP,82
X-Cache: MISS
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

No more data available.
Data was not received by the server.
parse_cmd() started
Received commands:
6
parse_cmd() completed
parse_receive_reply() completed
Can not send data, saving to flash memory
Sleeping for:
-2541
ms
Error: negative sleep time.
sms_check() started
Modem Reply:

AT+CMGL="REC UNREAD"

OK
Deleting READ and SENT SMS
Modem Reply:
AT+QMGDA="DEL READ"

OK

Modem Reply:
AT+QMGDA="DEL SENT"

OK

sms_check() completed
Ignition status:
0
Ignition is ON!
collect_all_data() started
gsm_get_time() started
Modem Reply:

Modem Reply:
AT+CCLK?

+CCLK: "30/07/15,05:07:57+00"

OK

Neville77, 2015-07-30:
gsm_get_time() result:
30/07/15,05:07:57+0
gsm_get_time() completed
$GPVTG,70.13,T,,M,0.01,N,0.02,K,D*0E
$GPGGA,050726.000,3800.7154,37*61
$GLGSV,2,2,07,67,29,215,29,75,23,148,33,83,01,205,17*50
$GNGLL,3800.7289,S,14515.9039,E,050758.000,A,D*58
$GNRMC,050759.000,A,3800.7289,S,14515.9039,E,0.01,70.13,300715,,,D*5A
GPS fix received.
Data is current.
Current time set from GPS time:
30/07/15,05:07:59+00
gsm_set_time() started
Modem Reply:
AT+CCLK="30/07/15,05:07:59+00"

OK

gsm_set_time() completed

GPS already available, breaking
collect_gps_data(): fix acquired
Failed checksums:
8
collect_all_data() completed
Current:
30/07/15,05:07:57+0[300715,5075900,-38.012150,145.265060,0.02,43.30,70.13,64,17]

gsm_send_at() started
Modem Reply:
AT

OK
AT

OK

gsm_send_at() completed
gsm_disconnect() started
Modem Reply:
AT+QIDEACT

DEACT OK

gsm_disconnect(): DEACT OK found
gsm_disconnect() completed
GPRS deactivated.
gsm_connect() started
Connecting to remote server...
0
Modem Reply:
AT+QIOPEN="TCP","updates.geolink.io","80"

OK

CONNECT OK

Connected to remote server: 
updates.geolink.io
gsm_connect() completed
gsm_send_http(): sending data.
30/07/15,05:07:57+0[300715,5075900,-38.012150,145.265060,0.02,43.30,70.13,64,17]

gsm_send_http(): Length of data packet:
119
gsm_send_http(): Length of header packet:
171
Modem Reply:
AT+QISEND=171

> 
gsm_validate_tcp() started.
Modem Reply:

SEND OK

gsm_validate_tcp() data delivered.
gsm_validate_tcp() completed.
gsm_send_http(): Sending IMEI and Key
Modem Reply:
AT+QISEND=38

> 
Modem Reply:

SEND OK

gsm_send_http(): Sending body
gsm_send_http(): Body packet size:
81
gsm_send_http(): Sending data chunk:
0
gsm_send_http(): chunk length:
81
gsm_send_http(): data sent.
parse_receive_reply() started
Modem Reply:
AT+QISEND=81

> 
SEND OK

+QIRDI: 0,1,0

Modem Reply:
AT+QIRD=0,1,0,100

+QIRD: 78.46.45.61:80,TCP,100
HTTP/1.1 200
Modem Reply:
AT+QIRD=0,1,0,100

+QIRD: 78.46.45.61:80,TCP,82
X-Cache: MISS
Modem Reply:
AT+QIRD=0,1,0,100

ERROR

No more data available.
Data was not received by the server.
parse_cmd() started
Received commands:
6
parse_cmd() completed
parse_receive_reply() completed
Can not send data, saving to flash memory
Sleeping for:
-2454

opensourcegps, 2015-07-30:
I believe the issue is related to the fact that OpenTracker sends data through http/1.0 connection goes through some proxy of you GSM provider and gets converted to http/1.1.

I will investigate further, probably the best solution could be to make OpenTracker to use http/1.1.

opensourcegps, 2015-08-14:
Closing as no response for 2 weeks. Please re-open if problem still persists.

closed 2015-08-14 by opensourcegps