geolink/opentracker #32: Code formatting reorganization

opensourcegps, 2015-08-12:
I would like to propose formatting current code style into single and consistent format. This affects indents, parenthesis style, etc. Current code was based on some non-standard style and it might be a good idea to switch it now to more common and comfortable version.

Once we choose most standard format, we stick with it for further development. Please propose format that will work for you, or just submit pull request with formatted files and we will accept it. 

It's important to choose such format that will be comfortable for the community and further development.

m4rkw, 2015-08-12:
2 character space indent is very popular I think.  I tend to prefer tabs but spaces seems more common, and also causes less issues if you have strings in the code that contain spaces.

opensourcegps, 2015-08-12:
agree with 2 character space indent

m4rkw, 2015-08-12:
Would also be nice to set standards for things like where the opening { is, on the same line or the next line.  Eg:

if (FOO) {

vs

if (FOO)
{

I think the latter tends towards more readability but it's really a personal choice thing.

opensourcegps, 2015-08-12:
if (FOO) { - I think this is more common. Even through I got used to second option. Maybe we can just use some IDE to auto format everything and make it standard?

m4rkw, 2015-08-12:
Most IDEs will have settings for how the indentation should work which people play around with, and programmers tend to be fussy about the tools they use (I for example mostly only edit code in vim).  I think if the community is going to be big it will inevitably encounter users who want to use their choice of editor, which should be fine as long as it's configured to the coding standards.

I'm happy with either if (FOO) {  or the other style, imho it's more important that it's consistent than which way we do it.

m4rkw, 2015-08-12:
Have reformatted the code with the following style guides:

- 2-character space indentation
- { is always on the same line as the block initiator rather than a new line
- no spaces between function names and opening (
- no trailing whitespace

Happy to change any aspect of this, as now that it's all consistent changes to the style guidelines should be pretty easy.

Once happy with it it's probably worth putting these into the README as a guide to potential contributors.

dimetron, 2015-08-12:
To make it simple to all who is new to the project, Arduino build in Auto-format Ctrl-T should be sufficient.
However, it doesn't solve  FOO{} issue.

What about Google style guide ? 
https://google-styleguide.googlecode.com/svn/trunk/cppguide.html

Or the one from Linux kernel ?
https://www.kernel.org/doc/Documentation/CodingStyle

[image]

dimetron, 2015-08-12:
Mark - do you agree on AutoFormat? Could you please apply/commit that also on top ?
This will prevent future formatting issue, of course, if everyone will do auto format before commit ;)

m4rkw, 2015-08-12:
To be honest I think the Arduino IDE is a pos and any use of it should be discouraged, but you're not wrong that it's probably what most people will end up using.  I don't have it to hand at the moment, but if running autoformat doesn't totally bork everything / make it unreadable then I'm not against doing it.  I would like to note however that I'm pretty sure a long of the indentation issues and trailing whitespace that were in the codebase are the result of people use that IDE.

dimetron, 2015-08-12:
I am not a big expert in cpp, but if you are saying ArduinoIDE is a pos, lets find alternative.

Then maybe some other tool like Astyle can help ?  It works under Win/Linux/OSX 
http://sourceforge.net/projects/astyle/?source=typ_redirect

Here is example after default format applied opentracker:
https://github.com/geolink/opentracker/compare/master...dimetron:master?diff=split&name=master

I guess one additional command to git would not make a big difference ?

dimetron, 2015-08-12:
Same topic on SOF - http://stackoverflow.com/questions/841075/best-c-code-formatter-beautifier

m4rkw, 2015-08-12:
Well, if you're talking about choosing a new IDE that everybody who works on opentracker should use, that's not going to work for reasons already stated (people like to use the tools they like).  If you're just talking about running it through a prettifier, I'm not sure it's necessary at this point - I already cleaned it up by hand to the point where changes to the formatting resulting from conscious decisions to do things a different way can probably be made with simple regex replaces.

opensourcegps, 2015-08-12:
I agree that we can't force people to use specific IDE or some prettifier. Let's just keep it simple, it's formatted now and we can put notice in Wiki what format we use and ask to use it too. Basically case closed.

The main reason for this is just that pull request does not have whole file modified, because somebody put 3 spaces instead of 2 on every line. Apart from that, everyone can use what ever tools they like. At the end it's not a big deal, if somebody puts some extra spaces on their own lines they commit, this is still not a problem. As long as the code works :)

closed 2015-08-14 by opensourcegps