geolink/opentracker #34: Detecting the engine running

m4rkw, 2015-08-13:
So the current ignition detection actually just detects that the key is at position 2, rather than the engine is actually running (because that's all we get from the ignition wire).  It's just occurred to me that you can easily see whether the engine is running by the battery level - if the ignition signal goes positive and the voltage dips then it's just the key at position 2.  If the signal goes live and the voltage increases then the engine is running.

Should be pretty easy to reliably code this into the software, I'll do it when I have time but am documenting this here for now.

fhasex20, 2015-08-13:
Battery voltage should be about 13,3-14,8 if engine is running, at 24volt Systems it May be 26-28volts. But the easiest thing would be to hook up can-bus and watch rpm's, in this case you could get almost any data from the bus. Last time i tried canbus wasn't working correctly, back in the Tigal daysSo far,Ferdinand                                                                                                                                                                                                                Von: Mark WadhamGesendet: Donnerstag, 13. August 2015 11:28An: geolink/opentrackerAntwort an: geolink/opentrackerBetreff: [opentracker] Detecting the engine running (#34)So the current ignition detection actually just detects that the key is at position 2, rather than the engine is actually running (because that's all we get from the ignition wire).  It's just occurred to me that you can easily see whether the engine is running by the battery level - if the ignition signal goes positive and the voltage dips then it's just the key at position 2.  If the signal goes live and the voltage increases then the engine is running.

Should be pretty easy to reliably code this into the software, I'll do it when I have time but am documenting this here for now.

m4rkw, 2015-08-13:
Yeah CAN-bus is on my todo list.. for some mythical future where I don't have a dayjob and am not about to have a kid :D  I'll get it to eventually

fhasex20, 2015-08-13:
Iam on it, will let you know if i success                                                                                                                                                                                                                                                                                                                                         Gesendet von meinem BlackBerry 10-Smartphone.                                                                                                                                                                                                                Von: Mark WadhamGesendet: Donnerstag, 13. August 2015 15:26An: geolink/opentrackerAntwort an: geolink/opentrackerCc: fhasex20Betreff: Re: [opentracker] Detecting the engine running (#34)Yeah CAN-bus is on my todo list.. for some mythical future where I don't have a dayjob and am not about to have a kid :D  I'll get it to eventually

closed 2016-07-01 by ppescher