geolink/opentracker #35: Programming on LInux

varlux, 2015-08-23:
I'm trying to install OpenTracker board in Arduino IDE (Linux 32bit) but, when i press "install" button i read a red message "Tool bossac is not available for your operating system" and procedure stops. Can somebody help me? Thanks

fhasex20, 2015-08-23:
You have to build bossac from source.Git Clone                                                                                                                                      https://github.com/geolink/opentracker-bossa.gitGit clone ‎ https://github.com/geolink/opentracker-arduino-board.gitEdit ‎local package_opentracker_index.jsonAnd remove bossac entry . Dirty, bu:t should work                                                                                                                                                                                                   Ferdinand Haselbacher                                                                                                                                                                                                                Von: varluxGesendet: Sonntag, 23. August 2015 19:13An: geolink/opentrackerAntwort an: geolink/opentrackerBetreff: [opentracker] Programming on LInux (#35)I'm trying to install OpenTracker board in Arduino IDE (Linux 32bit) but, when i press "install" button i read a red message "Tool bossac is not available for your operating system" and procedure stops. Can somebody help me? Thanks

salanki, 2015-09-11:
The same works for Mac. I however had to modify the opentracker-bossa Makefile:

COMMON_CXXFLAGS=-arch i386 -arch x86_64 -mmacosx-version-min=10.10 -stdlib=libc++ -std=c++11
COMMON_LDFLAGS=-arch i386 -arch x86_64 -mmacosx-version-min=10.10

and also install wxmac:
brew install wxmac

ppescher, 2015-09-13:
FYI, the board suport package now includes linux binaries (since v1.0.1)
See https://github.com/geolink/opentracker-arduino-board/commit/98105ba1d54bdc6c8949accd23c333a4d9dc58de

To build the command line tool, you don't actually need wxWidgets. Just "make bin/bossac"

ppescher, 2015-09-16:
Can anybody please test the Mac OSX binaries, before the official release?

You should be able to get v1.0.2 of the board support package using this URL in your preferences: https://github.com/geolink/opentracker-arduino-board/raw/1.0.2/package_opentracker_index.json

That includes the OSX binary of the BOSSA tool.

varlux, 2015-09-17:
Grazie Paolo, seguiro' il tuo consiglio e ti faro' sapere

varlux, 2015-09-28:
OK, it works on Linux, Thanks!

ppescher, 2015-09-29:
Thank you varlux.
Any Mac user that can test on that platform?

closed 2016-07-01 by ppescher