geolink/opentracker #37: Can't SPI

firstroboman, 2015-09-09:
I need to communicate with a SPI based sensor and tried including <SPI.h> in Opentracker_3_0_1 but now it won't build.

//External libraries
#include <SPI.h>
#include <TinyGPS.h>
#include <avr/dtostrf.h>
#include <DueFlashStorage.h>

Arduino IDE version is 1.6.5 and the board target is "Geolink OpenTracker"

Error messages are:
error: 'PIOC' was not declared in this scope
error: 'PIOD' was not declared in this scope

firstroboman, 2015-09-09:
Sorry .... some of the text was omitted from my initial post.  I'm trying to include SPI.h under the External libraries comment.

ppescher, 2015-09-09:
Just pushed a new commit/branch at https://github.com/geolink/opentracker-arduino-board/commit/393eb67861007427d8f100a8b9b99685363511e8 with a proposed fix.
 
You can get the relevant file from: https://github.com/geolink/opentracker-arduino-board/raw/1.0.2/1.0.2/libraries/SPI/SPI.cpp
 
And replace it on your folder:
%AppData%\Arduino15\packages\opentracker\hardware\sam\1.0.0\libraries\SPI
Or:
%AppData%\Arduino15\packages\opentracker\hardware\sam\1.0.1\libraries\SPI
 
Depending on what version you are using. The only difference between 1.0.0 and 1.0.1 is Linux binaries for BOSSAC (handling sketch upload).

If you confirm it works, I will make a new release of the board support package.

closed 2016-07-01 by ppescher