geolink/opentracker #39: Can't read I2C

adrfrank, 2015-09-15:
I need to integrate some i2c sensors to the opentracker device, when I try to request data from the these sensors nobody is responding.

I verified all connections for the i2c bus, including pull ups resistors, and all seems to be fine. I though it was a sensor problem, but this problem was present in  all sensors that I connected and only with the OpenTracker, because I tested these sensors with UNO and DUE and nothing was wrong.

I think the problem is in the variant for Due-OpenTracker but it is only a suspicion.

ppescher, 2015-09-17:
We verified I2C works as intended with an external EEPROM device. There are no known software or hardware problems.
Please check your connections and post the relevant code if you still have problems, along with details on your setup.

adrfrank, 2015-09-17:
I'm using the pins 3 and 4 from the j2 connector. The sensor is a MCP9808 and I tried to read using the arduino due library for this sensor.

What did you use to read the I2C bus? (I used wire.h)  Can you please post the setup for your i2c test?

ppescher, 2015-09-18:
```C++
#include <Wire.h>

#define Console SerialUSB
#define EEPROM_ADDR  0x50 // 24LC128 EEPROM I2C Device Address

void setup()
{
  delay(3000);
  
  Wire.begin();
  uint8_t data = 0x69; // Test data
  uint16_t addr = 0x1234; // Test address

  Wire.beginTransmission(EEPROM_ADDR); // Chosen base address
  Wire.write(addr >> 8);  // send MSB of the address
  Wire.write(addr & 0xFF); // send LSB of the address
  Wire.write(data); // send data
  Wire.endTransmission();
  
  delay(100);
  
  Wire.beginTransmission(EEPROM_ADDR); // Chosen base address
  Wire.write(addr >> 8);  // send MSB of the address
  Wire.write(addr & 0xFF); // send LSB of the address
  Wire.endTransmission(false); // don't generate a stop
  Wire.requestFrom(EEPROM_ADDR, 1, true); // Start read and stop at the end
  if (Wire.available() == 1) { // if one bytes was received
    Console.println(Wire.read());   // print the reading
  }
}
```
[image]

Please check your connections, make sure the sensor is powered at 3.3V and that you use the device 7-bit I2C address for communication with the Wire library.

adrfrank, 2015-10-12:
Thank you, the problem was present when I connect more than two I2C devices. And I resolved it to change the connector type.

closed 2015-10-12 by adrfrank