geolink/opentracker #42: gsm_get_reply()

agraciabona, 2015-12-14:
Hi,
I began to use opentracker with software release V2.0.1 and Arduino IDE 1.5.8. I modified that software to make my own and it works fine.

Now I tried to use my opentracker software with new IDE 1.6.x but I had problems in function gsm_get_reply(). The problem was that characters were lost when reading from gsm modem in parse_receive_reply(). I see that IDE 1.6.x comes with “#define SERIAL_BUFFER_SIZE 128” in RingBuffer.h and this is the problem. In V2.0.1 documentation in “Opentracker2_Manual_1.0.1.pdf” it says to write “#define SERIAL_BUFFER_SIZE 256”. My software only works fine with 256...

I think it is necessary to include in current documentation instructions to modify this file or supply a RingBuffer.h file with 256 not 128.

ppescher, 2015-12-14:
You don't need to patch your Arduino IDE manually anymore.
Please try to install the board support package from https://github.com/geolink/opentracker-arduino-board
It may be compatible with v2.0.1 of the software, or you can try v3.x from this repo.

agraciabona, 2015-12-31:
Ok I see now the support package https://github.com/geolink/opentracker-arduino-board comes with 256 in RingBuffer.h. Anyway, if there is a lot of data allways some characters are lost. Now I'm working with 2048 in SERIAL_BUFFER_SIZE in RingBuffer.h to avoid problems. I think the problem is in RingBuffer but I'm not sure. I'll try to find it out.

closed 2016-07-01 by ppescher