geolink/opentracker #45: GPS module reset 10-30s cycles

speeedy81, 2016-01-29:
I was wondering why my opentracker was loosing fix while driving and even stationary. When investigating more I noticed that GPS module reset without any reason.  Could it be because of the power source (voltage not stable) because with some power sources it does not happen ?

m4rkw, 2016-01-29:
When you say the GPS module resets, what do you mean exactly?  Are you seeing it lose the fix in the serial output and go through all the blurb again looking for a new fix?

agraciabona, 2016-01-29:
My opentracker resets sometimes (not allways) when turning on the engine. Only in this case. Driving, it doesnt happens. I think if voltage drops under a certain value it resets.

speeedy81, 2016-01-30:
Yes, I see gps losing the fix from serial data.

Here is caption of that data when that happens, tracker was powered at my desk from 12V battery so power source was very stable.

$GPVTG,90.42,T,,M,0.24,N,0.44,K,A*04
$GPGGA,114525.000,6130.5781,N,02336.0557,E,1,6,1.39,104.8,M,20.5,M,,*51
$GNGSA,A,3,09,23,07,02,16,,,,,,,,1.68,1.39,0.94*1F
$GNGSA,A,3,67,,,,,,,,,,,,1.68,1.39,0.94*14
$GPGSV,3,1,11,09,78,138,34,23,44,099,35,07,44,192,14,02,33,275,22*7D
$GPGSV,3,2,11,16,32,067,35,06,22,227,19,26,22,037,28,05,21,304,29*70
$GPGSV,3,3,11,30,15,206,14,29,14,343,23,40,12,135,*44
$GLGSV,1,1,01,67,33,126,32*51
$GNGLL,6130.5781,N,02336.0557,E,114525.000,A,A*4D
$GNRMC,114526.000,A,6130.5790,N,02336.0550,E,0.3$PMTK011,MTKGPS*08
$PMTK010,001*2E
$PMTK011,MTKGPS*08
$PMTK010,002*2D
$GPRMC,235943.799,V,,,,,0.00,0.00,050180,,,N*4C
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,235943.799,,,,,0,0,,,M,,M,,*45
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPGSV,1,1,00*79
$GLGSV,1,1,00*65
$GPGLL,,,,,235943.799,V,N*77
$GPRMC,235944.799,V,,,,,0.00,0.00,050180,,,N*4B

closed 2016-07-01 by ppescher