geolink/opentracker #46: Tracker hangs

agraciabona, 2016-02-01:
My opentracker is working 24h a day, it is allways powered on. About every 3 days it hangs. Sometimes it hangs twice in a day and other times it is 5 days working without problems. When it hangs I realise of it because It stops trasmiting data and if I connect the serial cable to it, my computer says "usb device not recognized". Power off and on and tracker works again. It looks like an electrical problem, not a software problem. I installed two diodes (one in 'power' and the other one in 'ignition') to avoid voltage peaks but it still happens. Has someone experienced something similar?

ZdenekAster, 2016-02-01:
I have 5 opentrackers in truck. All opentrackers have the same software. Three of them is work without problems. But two have the same problems like yours....

agraciabona, 2016-02-04:
ZdenekAster, your opentrackers are allways in the same trucks? Or have you made the test of changing one that fails and put it in other truck?

ZdenekAster, 2016-03-04:
No, I have 5 truck. In every truck one opentracker. In generally Opentracker hangs at the same place. Near border France/Spain, near city Lyon and etc. I test restart GSM module before every communication. So far, it looks good, but I test only one week.

m4rkw, 2016-03-04:
@ZdenekAster Next time you go that way I'd suggest having a computer connected to the serial output and see what point it hangs at.  Check this a few times and if it's always in the same point in the run loop it may help with debugging it.

m4rkw, 2016-03-04:
@ZdenekAster If you can run this test 2-3 times and send me the serial buffers for each test up to the point where it hangs (don't need the entire thing, just one or two complete loops before the hang) I'll look into it for you.

ZdenekAster, 2016-03-04:
That is difficulty. On my opentracker on my desk work correct whole month. This problem is in the truck on the road, it is not possible travell with truck and notebook and grab serial output for debug.

m4rkw, 2016-03-04:
@ZdenekAster appreciate it's difficult but if it is a software glitch that's the only way we'd be able to debug it, otherwise there's nothing to go on.  The fact that it seems to be happening in the same location for you every time suggests there may be some set of circumstances happening that the tracker somehow can't deal with.  Without seeing where it gets stuck we have no clues at all for debugging.  Maybe if you have a spare tracker you could connect it to the 12V socket so it could just sit on the passenger seat with a laptop and the aerial on the dash?  Just an idea, hope you get it sorted anyway!  Mark

agraciabona, 2016-03-04:
I've been working on this issue. I think there are two kind of hangs:

A.- It seems a hardware problem: when you connect usb cable the computer shows 'usb device not recognized'. I have no idea of what to do in this case.

B.- It seems a software problem: sometimes, when it tries to connect it allways gets IP_INITIAL, allways! forever! even disconnecting and connecting again. This happens with my car parked in my house: I see tracker working during hours and suddenly it stops trasmiting. The only way to get out of it is doing a reset. I've programed a reset after several connect failures and I It solves the problem.

Problem 'B' happens many more times than 'A', in fact I've seen 'A' only two times. So, I think ZdenekAster  is having problem 'B' because he says he solves it doing a gsm restart.

m4rkw, 2016-03-06:
Interesting.  I've had A once but it only froze when I plugged the usb cable into a mac, so I assumed there was some kind of driver glitch.  I wired a push-to-break reset switch in so I could reset it if it happens again.  Haven't seen B though, my tracker has been working for a long time without freezing up.

I'm guessing you're using the latest code from the master branch, so something about your situation must be different from mine and the others who don't see the problem.  Could be network related maybe, have you tried using a different network to see if that makes a difference?  A quick google suggests that "IP INITIAL" isn't an error message, rather it indicates a state where the IP stack is initialised.  Could you post the AT commands being sent before and after it gets stuck at this point, ideally the full serial output?  One of the AT commands must be triggering a hang and it'd be interesting to see which one.  Mark

agraciabona, 2016-03-07:
I'm not using last code from master branch, I modified myself but the communication procedure is the same. I know IP_INITIAL is not an error but how is it possible it gets stuck on it? disconnecting and connecting forever and allways getting IP_INITIAL? And, if opentracker issue a reset it starts comunication inmediatly... This happens without moving the tracker, so gsm coverage doesn't change. I don't have the serial output when the problem begins, but I suppose that after a correct communication the next one comes with the problem. Anyway, this is not a big problem for me, now I can get rid of it issuing a reset and I only lose communication during 2 or 3 minutes. 
To me the other hang ('A' type) is worse because it is impossible to program anything to avoid it, the tracker seems electrically 'dead'. Maybe it only happens when the tracker is onboard a vehicle powered by the vehicle battery because voltage is unstable. Anyway, I found a way to get rid of it: I put a diode in the power supply cable to limit voltage to 12V. Doing this, when the car driver starts the engine, power supply falls so much so the tracker resets because low voltage :-) and if it was hanged it comes to life again. 
So, now I can avoid all kind of hangs, or I hope so... I'll keep you informed.
One question for @ZdenekAster: what happens when your tracker hangs? You loose communication with your truck until it comes next to you and you reset it?

ZdenekAster, 2016-03-07:
2 agraciabona: When tracker hangs, loose communication. if someone connect/dissconnect main connector (power off/on ) tracker normaly work. I think that problem to registration to GSM network in roaming. It is difficult to repeat this problem because, that problem occurs twice a month at home on desk track work month without problem.
I design pcb with atxmega and sd-card for expansion connector on tracker for log debug communication, but currently i do not have time for develop this :-(

agraciabona, 2016-03-07:
If it is a problem in registration to GSM network in roaming, it should work again when returning to home country? My taxi whent once to France (I live in Barcelona) and it lose comunication in France, but when returning to Spaint it worked again.

ZdenekAster, 2016-03-07:
No, only power reset solve this problem. Truck ride from Czech (home country), Germany, France, Spain and back.. Track hangs very often near France/Lyon and Spain/La Jonquera...

ZdenekAster, 2016-03-11:
Today, I get report from my prototype logger with problem hangs. Modem reply ERROR on command AT+QIOPEN....
[logfile.txt](https://github.com/geolink/opentracker/files/169918/logfile.txt)

agraciabona, 2016-05-13:
I haven't seen again hang type 'A'. I put in my car a bigger battery, maybe this is the reason...
Type 'B' stills happens but I modified software to issue a reset if communication is lost during 3 minutes and this is the way to make it work again.

ppescher, 2016-09-14:
The hardware issue might be related to the presence of a resettable fuse on-board, in series with the power supply. With higher current consumptions or higher temperatures it may lead to an excessive voltage drop (due to increased resistance), and even cause device resets with low input voltages (or improper power cycling that cause device lock-out).
The resettable fuse has been removed since board revision 2.4. An external in-line fuse should be used instead.

closed 2017-04-05 by ppescher