geolink/opentracker #52: Mac Yosemite I don't see usb port

oscaroxy, 2016-05-08:
I've Mac Yosemite, I tried to install "FTDI chip"
`http://www.mommosoft.com/blog/2014/10/24/ftdi-chip-and-os-x-10-10/`

also if when I run 
`sudo kextload /System/Library/Extensions/FTDIUSBSerialDriver.kext/`

I don't receive nothing result with command
`ls /dev |grep usbserial`

I should see 

`cu.usbserial-A9GBBDLL`
`tty.usbserial-A9GBBDLL`
 
without this I don't see port in arduino IDE, so I can't program the opentracker... :-(

Can anybody help me?
thanks

ppescher, 2016-05-08:
Hello,
There is no FTDI chip in OpenTracker. It works the same way as for Arduino DUE.

For Windows you need to install drivers that are provided with Arduino IDE installation.
For Linux/MacOSX I'm not sure, but I think you don't need any additional driver. You should see the serial port listed in the Arduino IDE menu. It may be "usbmodem" instead of "usbserial".

That of course only if the board is running correctly.

oscaroxy, 2016-05-10:
I am going crazy, I tried everything but nothing, my PC don't see the board ... I tried also by windows 8.1, don't only by Mac OSX... 
In attach I put I see... the leds blink so power there is... the usb is connected... the board is installed into Arduino IDE, look bottom in image for to see opentracker board (in preference I put the json)... 
the only thing I didn't do is the SIM, but I don't believe that I need SIM because the PC don't see 

[opentracker_img.zip](https://github.com/geolink/opentracker/files/257291/opentracker_img.zip)

ppescher, 2016-05-10:
Please do your tests on Windows before you move to MacOSX.
When the USB cable is plugged in at both ends, you should see the board as a new serial port, as in the following screenshot:
[image]

The board can work without USB cable, it is not powered by USB.
If you don't see the virtual serial port, please first check the USB cable. Try to use the same cable with something else, for example to copy any file from your smartphone to your PC. Or try to use another USB cable with the OpenTracker, sometimes the connector gets loose and makes poor or no electrical contact.

oscaroxy, 2016-05-10:
You are my hero and I'm stupid ... It was the usb cable, now it work fine both Windows and Mac... I'm sorry and thanks a lot in order to patience.
Your product is wonderful :-) I'm happy in order to have buy it

Thanks 

Inviato da iPhone 6

closed 2016-05-11 by oscaroxy