geolink/opentracker #53: Method name - openPort(); Exception type - Port busy

oscaroxy, 2016-05-15:
I don't understand why when I uplaod 2 sketch the second obtain this error:
`processing.app.debug.RunnerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:141)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.Sketch.upload(Sketch.java:1186)
	at processing.app.Sketch.exportApplet(Sketch.java:1160)
	at processing.app.Sketch.exportApplet(Sketch.java:1132)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2381)
	at java.lang.Thread.run(Thread.java:745)
Caused by: processing.app.SerialException: Errore durante l'esecuzione del comando touch sulla porta seriale '/dev/cu.usbmodem1411'
	at processing.app.Serial.touchForCDCReset(Serial.java:87)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:130)
	... 6 more
Caused by: jssc.SerialPortException: Port name - /dev/cu.usbmodem1411; Method name - openPort(); Exception type - Port busy.
	at jssc.SerialPort.openPort(SerialPort.java:164)
	at processing.app.Serial.touchForCDCReset(Serial.java:81)
	... 7 more
`

I close the Serial Monitor, after I close also the IDE arduino, but nothing the error is port busy...
I have to turn-off the power to board and after tunr-on it... 
thanks

oscaroxy, 2016-05-16:
How I resolve this? 
This is invalidate for me, every time I have to turn-off and after turn-on the power ... 
thanks

ZdenekAster, 2016-05-16:
If you try unconnect and connect usb cable, resolve this problem too?

oscaroxy, 2016-05-16:
sometime, no always ... but why does this? is there a way in order to solve this?
How do you program it?

ppescher, 2016-05-17:
Looks like it may be an issue related to MacOSX.
If you have Windows, please try the same procedure and see if it works for you.

oscaroxy, 2016-05-18:
Today I tried with virtualbox, but the usb don't work, it always is busy... I'll try with windows pc... 
in the meantime thanks

oscaroxy, 2016-06-02:
Hi, I would to use gsm.h library so I can to use the example of arduino. But When I include GSM.h and I compile receive the error:
WARNING: the GSM library claims to work on [avr] architectures and may not be compatible with your card that uses the architecture [sam]...
thanks _______________________________________________________________________
Oscar OXY

ppescher, 2016-06-02:
If that is a warning you should be able to continue and still use the library.
You should complain with the GSM library authors about this warning message.
There is nothing we can do.

closed 2016-06-02 by ppescher

oscaroxy, 2016-06-03:
Hi, I would to use gsm.h library so I can to use the example of arduino. But When I include GSM.h and I compile receive the error:

/Applications/Arduino.app/Contents/Java/libraries/GSM/src/GSM3SoftSerial.cpp:129:2: error: #error This version of GSM3SoftSerial supports only 20, 16 and 8MHz processors
 #error This version of GSM3SoftSerial supports only 20, 16 and 8MHz processors

other the warning:

WARNING: the GSM library claims to work on [avr] architectures and may not be compatible with your card that uses the architecture [sam]...

thanks for your patience 

ps: is opentracker 100% compatible with arduino? Why this base library don't work fine?

ZdenekAster, 2016-06-03:
Opentracker is compatible with Arduino DUE but use cpu with less ram and pins. 
Arduino due (ARM 32bit ) have different architecture from arduino uno (AVR 8bit)
GSM library is different for GSM chip, SIM900, QUECTEL M95, base commands are same, but ip stack is different. Your error use gsm wich use library software serial. Software serial most often on arduino uno... You can use gsm library from opent tracker.

oscaroxy, 2016-06-03:
thanks for your complete reply, but where is an example of gsm library from open tracker? I would to do a simple call http (gprs datas/datas traffic), that is I would send lat & long to my server (send to https://myserver.com?lat=41.021221&log=1.234495959)

thanks a lot

ZdenekAster, 2016-06-03:
Yes gsm is in gsm.ino. Easiest way is modify tracker.h, where is define hostname and include information to send, but opentracker use POST data define here https://geolink.io/configure.php,

JuniorEC, 2017-08-23:
Hello everyone, make sure there is no other program using or using an indirect way of communication, in case you have to use this port, create a rule to use one at a time.