geolink/opentracker #54: Serial.println don't work

oscaroxy, 2016-05-15:
Why  Serial.println don't work than debug.println yes and why if I do:

`
void setup() {  
  
  //setup led pin  
  pinMode(PIN_POWER_LED, OUTPUT); // Set LED as Output  
  digitalWrite(PIN_POWER_LED, LOW); // Set LED initially off  
  Serial.begin(9600);
   debug_print("setup");
   Serial.println("Serial setup");
}  
`

none string are print into monitor serial, why?
thanks

oscaroxy, 2016-05-15:
maybe I have to install: SoftwareSerial ?
Where Do I find that library and How Do I install it?

thanks

ZdenekAster, 2016-05-15:
In main program is in
#define gps_port Serial1
#define debug_port SerialUSB
#define gsm_port Serial2 

You have to write
SerialUSB.begin(9600)
debug_print("Setup);  // or SerialUSB.println("Setup");

oscaroxy, 2016-05-15:
Unfortunately it don't work, I done:

`int ii=0;
void setup()
{  
  debug_port.begin(9600);
  ii++;
  debug_port.println(ii );

  pinMode(PIN_POWER_LED, OUTPUT); // Set LED as Output  
  digitalWrite(PIN_POWER_LED, LOW); // Set LED initially on
}

void loop(){
 ii++;
  debug_port.println(ii );
}
`

the led turn-on, but in serial monitor I see the ii=4,ii=5,ii=6,etc... but it don't start from 1.... why?????
thanks in order to patience :-)

ZdenekAster, 2016-05-15:
Try command delay(2000);  before debug_port.being(9600);

oscaroxy, 2016-05-16:
yes, so work, because slow down ... this mean that the usb reply a message slow than the setup procedure.
The arduino uno reply the Serial.print in setup procedure correctly, why that board no?
thanks

ZdenekAster, 2016-05-16:
Open track is the same as Arduino due, is different from uno, This boards have two usb modes. I think  when change mode after programming to serial,  windows lost few first serial chars. Try less delay for work. It think that 500ms be enough

closed 2016-05-16 by oscaroxy