geolink/opentracker #55: Board version 1.0.3 fixes Serial3 pin definition, but not appearing in IDE

VernonBreet, 2016-06-21:
There was a problem with the pin definitions for Serial3, for which a new URL for board manager, was provided. I can't see the version 1.0.3 fix in the IDE though, and I can now not compile previously working code.

ppescher, 2016-06-21:
Branch 1.0.3 of https://github.com/geolink/opentracker-arduino-board provides support for the next board hardware revision 2.4
You have an option in the IDE Tools menu to choose your board revision.
Even when you choose revision 2.3 (and earlier) the old sketches may have build errors, which have been addressed in current master branch of https://github.com/geolink/opentracker

If you want to keep your old sketch, you can patch version 1.0.2 of the board support package manually to fix Serial3.
Open the file "variant.cpp", that you can find in "%LocalAppData%\Local\Arduino15\packages\opentracker\hardware\sam\1.0.3\variants\geolink_opentracker" on your PC (if that doesn't work, try %AppData%).
And replace **PIO_PERIPH_B** with **PIO_PERIPH_A** at lines 202-203, as below:
```
  // 42 - USART3 (Serial3) all pins
  { PIOB, PIO_PB20A_TXD2|PIO_PB21A_RXD2, ID_PIOB, PIO_PERIPH_A, PIO_DEFAULT, (PIN_ATTR_DIGITAL|PIN_ATTR_COMBO), NO_ADC, NO_ADC, NOT_ON_PWM, NOT_ON_TIMER },
```

VernonBreet, 2016-06-21:
Thank ypu, I have re-done everything, but still get an error, even when I select board version 2.3:
[image]
I will try a manual patch next, and let you know.

ppescher, 2016-06-21:
That error might be caused by the old version 1.0.2 still present on disk.
The IDE sometimes fails to delete the folder that contains the old version (try to remove "%LocalAppData%/Arduino15/packages/opentracker/hardware/sam/1.0.2" manually)

VernonBreet, 2016-06-21:
The manual patch worked, thank you.

closed 2016-06-21 by VernonBreet