geolink/opentracker #56: Arduino IDE integration files for 1.0.3

Daniel-dk, 2016-06-30:
Following the IDE integration instructions from https://github.com/geolink/opentracker-arduino-board which points to the master branch and  only lists up to 1.0.2

[image]

when installing 1.0.3 by directly [referencing its json file](https://raw.githubusercontent.com/geolink/opentracker-arduino-board/1.0.3/package_opentracker_index.json) these seems to be some information missing and compilation fails.
 
[image]
[image]
[image]

ppescher, 2016-06-30:
Make sure you uninstalled the old version.
Check if you have some left-over from the previous version in your folder: %LocalAppData%\Arduino15\packages\opentracker (on Windows)
In particular, under "opentracker\hardware\sam" there should be only one folder named "1.0.3"

FYI: The new version will be soon merged into the master branch. That should facilitate upgrading (but I've seen some IDE versions still failing to clean up properly).

Daniel-dk, 2016-06-30:
That was the issue, 
There were some directories left over in 1.0.2 with "NUL" files in them - probably form an incomplete removal.

after removing 1.0.2 from the arduino packages the IDE successfully uses 1.0.3

closed 2016-06-30 by Daniel-dk