geolink/opentracker #57: Cant't work with 1-wire on internal extension connector

varlux, 2016-07-03:
Hello, what is the name of the pins on the internal extension connector? I cant't communicate with my 1-wire sensor on "PIN_EXT_PA23".

Thanks

Daniel-dk, 2016-07-04:
Do you have a gist / pastebin where we could look at you code or test it on our boards ?

varlux, 2016-07-04:
Oh....yes...very simple code

`#include <OneWire.h>

OneWire ow(PIN_EXT_PA23);

byte owaddr[8];

void setup() {
  SerialUSB.begin(9600);
}

void loop() {

  if (ow.search(owaddr)) {
    SerialUSB.println(F("OK, I can found it!"));
    ow.reset_search();
  }
  else {
    SerialUSB.println(F("mmmm...no, it isn't working"));
  }
  delay(1000);
}`

Thank you Daniel

Daniel-dk, 2016-07-05:
`#include OneWire.h` should probably be `#include <OneWire.h>`
I could still not get your code ( or the oneWire library ) to flip the pin.. maybe there is some more setup needed that Dallas temperature does for us ?

I had some luck with the DallasTemperature library but I think my DS18B20 sensor is fried.. so one can only see the bits flipping but no data.
one can see the pin toggling, but no reply from the DS18B20- I think its fried, but I don't have another to test with..

[image]

I used this code : http://pastebin.com/6c4aYkck

varlux, 2016-07-05:
Hi Daniel, 

     thanks for your time. I thing I am inaccurate......excuse me.

The library I use is a [PaulStoffregen's library](https://github.com/PaulStoffregen/OneWire) create for Arduino IDE, and the simple code I wrote works fine in Arduino UNO.

Why this library doesn't work on OpenTracker?!?

Thanks

ppescher, 2016-07-05:
Probably due to this line:
https://github.com/PaulStoffregen/OneWire/blob/master/OneWire.h#L94

And possibly other places where only SAM3X8E is considered.
OpenTracker uses SAM3A4C (and SAM3A8C in early prototypes).

PaulStoffregen, 2016-07-06:
I do not have this hardware, and even if I did have it here, I probably could not spend engineering time to test it.

Please feel free to investigate this problem. When/if you find a solution (even just adding to the ifdef check), submit a pull request. Your changes should be limited to the hardware defines in OneWire.h.

This is how open source works. The OneWire library is designed to be adapted to other hardware by editing just those defines. As the maintainer of this library, I depend on community contributions to add the defines for hardware I do not personally own or support.

ppescher, 2016-07-06:
@varlux I guess you can just change that line from:
```
#elif defined(__SAM3X8E__)
```
To:
```
#elif defined(__SAM3X8E__) || defined(__SAM3A8C__) || defined(__SAM3A4C__)
```
They are essentially the same hardware family, often referenced as [SAM3XA](http://asf.atmel.com/docs/3.0.1/sam.drivers.spi.example.sam3x_ek/html/sam3xa_8h_source.html), so I think it should work. But I don't have OneWire devices to test it.

PaulStoffregen, 2016-07-06:
http://www.digikey.com/product-detail/en/maxim-integrated/DS18B20-/DS18B20--ND/956983

PaulStoffregen, 2016-07-06:
Please only submit a pull request *after* you have verified it works on your hardware.

ppescher, 2016-07-07:
@varlux I tested the patched OneWire library with OpenTracker HW rev 2.3 and a Dallas/Maxim temperature sensor chip DS18S20, powered at 3.3V and attached to pin PA23 with 4k7 pull-up resistor. It works fine.

varlux, 2016-07-07:
Grande Paoluccio!

I will test the patched OneWire library with my old Tigal OpenTracker and chip DS18B20 very soon.

varlux, 2016-07-21:
OK. I tried the patched library with old Tigal Opentracker (SAM3A4C).
It works fine.
Thank you all for your valuable time.

closed 2016-07-21 by varlux