geolink/opentracker #60: for inspiration update firmware over http

ZdenekAster, 2016-07-17:
I do not have more time for comment and create clearly code, but it is good function.
It would be a shame to share it. Sorry for my English.

On web is script which add bootloader (folder uflash) to binary my software for opentracker( Carunit_1_1), this script send binary as more block hex.
Opentracker get http get for every page and store over 128kB when get all change boot from section 1 and reboot. in section 1 now is downloaded bootloder with software bootloader reflash section 0 and change boot to section 0.

In carunit is interesting function wich set gprs parametrs from operator network code.
And function for get credit from prepaid sim card.

Software attached it is not complete because complex and web code for view trackers, but for include firmware update over http for own application is all that you need.
Software assume correct DueFlashStorage for access to both bank wich last parameter select bank.
[Export.zip](https://github.com/geolink/opentracker/files/367910/Export.zip)

Zdenek Aster

VernonBreet, 2016-07-18:
Good job, Zdenek! Busy checking it out now.

ZdenekAster, 2016-07-18:
I forget modified files opentracker define, where is change of linker script address for bootloader to boot section 1. In this attachment.
[opentracker.zip](https://github.com/geolink/opentracker/files/369135/opentracker.zip)
In arduino displaay Opentracker and Opentracker2 with modifed linker. I modified version 1.0.2 replace define file with this attachment, for newer version it probably will not work.

varlux, 2016-11-12:
Wow..... where I can find information?

ZdenekAster, 2016-11-12:
In first post is file export.zip, This code assume web support, but complete idea is in code. The code is cut from my project, but some notes and variables in czech, when i will have more time i am try tu rewrite clearly code, but for programmer is not problem use and modify for your application. In export.zip is all what you need.

varlux, 2016-11-19:
I am not a programmer, but I am very interested to this function. So, I will try learn it....with your help. Firstly I have to understand the procedure to reset  and erase the flash, and then how to program it "OTA".
By Arduino DUE's web page I know that is necessary to erase the flash before programm it, because in this case the board is started with the bootloader in the ROM memory.
Probably I have to use "erase" PIN, but this PIN is not connected to any others in openTracker.....
So, the first question is: How can I erase the flash in OpenTracker from a remote application? Which pins I have to use?

Thanks in advance

varlux, 2016-11-20:
Ok...ok... I am not a programmer and I need to understand "step by step".
I am studying your code, openTracker's pcb and processor datasheet......
So I understand that you firstly download new code (and "bootloader") in flash1, select boot from flash1 and reboot by C_REBOOT pin.
After reboot, the new "bootloader" in flash1 copies new code in flash0, select boot from flash0 and reboot.
Now firmware is update.

IS IT CORRECT ??

If so, I would like to ask you if I can test this procedure on my desk via serial port......
In detail, I want write a script that accepts data from serial port. After a specific command, I will send a compiled file. The script will copies this file in flash1 byte-to-byte thanks to DueFlashStorage library, select boot from flash1 and reboot.....

Can it works???

varlux, 2016-11-20:
Ops.... another question: where is C_REBOOT pin in version 2.4 ?!?