geolink/opentracker #62: Can't add OpenTracker Board V1.0.3 with board manager (Arduino IDE)

varlux, 2016-08-02:
These are the errors when I try to add the board:
----------------------------------------------------------------------------------------------------------------------------
Errore durante lo scaricamento di https://github.com/geolink/opentracker-bossa/releases/download/1.5-opentracker/bossac-1.5-opentracker-x86_64-pc-linux-gnu.tar.gz
java.lang.RuntimeException: java.lang.Exception: Errore durante lo scaricamento di https://github.com/geolink/opentracker-bossa/releases/download/1.5-opentracker/bossac-1.5-opentracker-x86_64-pc-linux-gnu.tar.gz
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$20(ContributionManagerUI.java:176)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: Errore durante lo scaricamento di https://github.com/geolink/opentracker-bossa/releases/download/1.5-opentracker/bossac-1.5-opentracker-x86_64-pc-linux-gnu.tar.gz
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:113)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:67)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:111)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$20(ContributionManagerUI.java:173)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 404
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:194)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:111)
	... 4 more
-------------------------------------------------------------------------------------------------------------------------------
Give me a hint, please!
Thanks

ppescher, 2016-08-03:
There was a typo in the download URL (404) in package index. It should be fixed now in the master branch.

varlux, 2016-08-03:
OK, it works.

Thank you Paolo

closed 2016-08-03 by varlux