geolink/opentracker #63: Use DueFlashStorage library with your board

varlux, 2016-08-26:
I saw the use of DueFlashStorage in your sketch.
Good.
I already use this library in my project, but I have a doubt about its management.... and I can't find out complete documents that helps me.

So I hope in your help...
In my project I want to store lots of configuration data, and my sketch is very big.
So...I am very scared....because I don't know if my configuration overwrite the code or if the code overwrite configuration data.

Are there struments to check it?!?

Thanks

ZdenekAster, 2016-08-26:
Opentracker use first 128 kB FLASH for code and second 128 kB FLASH for data, library DueFlashStorage write to second 128 kB FLASH. When You use more then 128 kB FLASH for code, you must add this size to addres to write.

varlux, 2016-08-26:
Ohh....thanks, so in actual configuration can I store 128 KB by DueFlashStorage library? And what about variables? Are they manipulate in SRAM?

varlux, 2016-08-29:
Another question......the first byte for DueFlashStorage is "0" in second bank of flash?!

I am confused.....

ZdenekAster, 2016-08-29:
Yes first byte for DueFlashStore = first byte in second bank of flash = position 0x20000 in all flash range.

varlux, 2016-08-29:
Ok....but please let me understand it. If I need more flash for my code, for example 200 KB, how do I teach my mcu and the library? With bossa? With configuration file? Or is it impossible?
Thanks for your time

ZdenekAster, 2016-08-29:
You can use library with from value 73728, its 131072+73728=204800. And size of code when you use opentracker you have to modify boards.txt (folder depends on operating system follow \packages\opentracker\hardware\sam\1.0.3 and linker script flash.ld

varlux, 2016-08-30:
mmmm....
I want to test myself.....please check my statements:

If I need 200 KB for my code in Opentracker, I first have to modify file boards.txt and change the line 9 from:

geolink_opentracker.upload.maximum_size=131072

to:

geolink_opentracker.upload.maximum_size=204800

Then I have to modify the file linker_scripts/flash.ld and change the lines 37-38 from:

rom (rx)    : ORIGIN = 0x00080000, LENGTH = 0x00020000 /* Flash, 128K */
eeprom (rx) : ORIGIN = 0x000A0000, LENGTH = 0x00020000 /* Flash, 128K */

to:

rom (rx)    : ORIGIN = 0x00080000, LENGTH = 0x000000C8 /* Flash, 128K */
eeprom (rx) : ORIGIN = 0x000002C8, LENGTH = 0x00000038 /* Flash, 128K */

From this, DueFlashLibrary will work correctly and it will store data in 56 KB.

Is it correct?
Thanks

ZdenekAster, 2016-08-31:
This values in byte not kilobyte!!! You modify 0x00020000 = 128k
to C8=200bytes......

rom (rx) : ORIGIN = 0x00080000, LENGTH = 0x00032000 /* Flash, 200K /
eeprom (rx) : ORIGIN = 0x000B2000, LENGTH = 0x0000E000 / Flash, 56K */

But DueFlash does not accept this. Library write from 88000, you need add offset 73728 to write correct place in flash....

Datasheet cpu in opentracker http://www.atmel.com/Images/Atmel-11057-32-bit-Cortex-M3-Microcontroller-SAM3X-SAM3A_Datasheet.pdf

varlux, 2016-08-31:
Oh my God......sorry for the error. :)
I think you are starting to hate me, but I hope you understand........ I chose OpenTracker because fully customizable. So, if I don't understand it all, then I understand nothing.

I know, DueFlashStorage write from A0000 because patched for OpenTracker, but if I need to customize it WHERE I need to add offset? 

Thanks for your patience

ZdenekAster, 2016-09-01:
You can modify DueFlashStorage.h line #define  FLASH_START  ((byte *)IFLASH1_ADDR) to
#define  FLASH_START  ((byte *)(IFLASH1_ADDR+73728))
Or modify function in this library...

varlux, 2016-09-04:
Thank you professor for your lessons, but now I have another doubt: if DueFlashStorage needs to modify its file DueFlashStorage.h (#define FLASH_START ((byte *)(IFLASH1_ADDR+73728))), WHY modify the file linker_scripts/flash.ld also?

Changes in this file are related to library, or to size of the code?

Thanks

ZdenekAster, 2016-09-04:
flash.ld is information for linker, when linker get over the size of flash, he give an error, it is not necessary...
the normaly all whole flash is for code, in script linker i give information for linker maximum length of code. DueFlashStore it is the same problem, when you not modify this, you risk rewrite your code with data and crash...

varlux, 2016-09-05:
Then....I think that to write data over the address 0x20000, I simply add offset in the file DueFlashStorage.h:

#define FLASH_START ((byte *)(IFLASH1_ADDR+offset))

and....If I need to write code over 128KB, I simply change the line 9 of the file board.txt:

geolink_opentracker.upload.maximum_size=max_byte_of_the_code

Are these steps enough for my goal?
And...sorry for ignorance, what is linker?

ZdenekAster, 2016-09-05:
It is ok, and linker flash.ld you must modify too, because linker use only space in this define.

closed 2017-04-05 by ppescher