geolink/opentracker #64: upload issues ( USB ) when device it attached to EXT_RX and EXT_TX

Daniel-dk, 2016-09-05:
we have an expansion shield on the Opentracker ( rev 2.3 ) with a device on EXT_RX and EXT_TX ( Arduino "Serial" )

I know the bootloader listens on the UART0 ( Arduino "Serial" / EXT_TX ,EXT_RX ) and USB for new code, when we have a device is attached to those pins it seems to interfere with the upload process even if only USB is used.

The board  "gets stuck" and we have to detach the shield, re-do the flash erase with the ERASE pin, upload and re-attach the shield.

do you know how we could get around this ?

I think this is more a SAMBA issue than a OpenTracker issue but useful to know about for anyone that wants to attach other serial devices to OpenTracker 

( rather use EXTRDX2 and EXT_TXD2 - Arduino "Serial3"

haxyn, 2016-09-06:
Hi,

Not sure if it's related, but I have similar problem.
I use EXT_SDA and EXT_SCL for I2C and EXT_21-23 for encoder.
Work perfect on Win7Pro, but on other machine with Win10Pro I have to use ERASE pin every second upload.

ppescher, 2016-09-06:
@Daniel-dk If something is connected to the boot UART, you should make sure that the serial line is idle at reset.

When the Arduino IDE opens the USB-Serial port at 1200 baud, then closes the port, the Arduino core libraries running on the board will reset the device to enter the ROM bootloader (SAM-BA).
This code will check both USB and UART in a loop, to see which one is connected first:
- if USB enumeration starts (connection to a PC) then the USB channel will be used
- if anything is received on the UART, then only that channel will be used

So make sure pin ETX_RX is always high (idle) at reset. If the rise time of this line is too slow, you might need to delay the nRESET rising edge, so that EXT_RX has time to reach the idle state before the boot program starts.

Please note the EXT_RX line should stay idle until the USB enumeration, which may take some time since reset (hundreds of milliseconds, or more, depending on the PC). If you have some other device that starts transmitting on the EXT_RX line, make sure it will delay the first transmission by the appropriate amount of time.

@haxyn I don't think this is related to your problem. The pins you are using should not conflict with the boot monitor.

Hope this helps,
Paolo

closed 2016-09-12 by Daniel-dk