geolink/opentracker #66: 3G Board - Failed auto-baudrate & +CME Error: 4

alskyline, 2017-02-03:
Here in USA, some of the major carriers are dropping their 2G service. This forced me to upgrade to the new 3G boards. The modem is having trouble connecting to the cellular network. I have attached a .txt file with the debug output. I have switched cellular carriers and my old 2G Opentrackers connect just fine. This seems to be only a problem with the new 3G boards.

Please Help.

[Debug output.txt](https://github.com/geolink/opentracker/files/750528/Debug.output.txt)

ppescher, 2017-02-03:
Are you using factory programmed firmware or did you re-program the boards yourself?

If you reprogram the boards with the firmware available here on GitHub, make sure you have the latest board support package from https://github.com/geolink/opentracker-arduino-board and you select Board Version: OpenTracker 3G in the Tools menu of your Arduino IDE.

Please also note the last IDE version known to work with OpenTracker is 1.6.13 at this date.

ppescher, 2017-04-05:
I have tested Arduino IDE 1.8.1 with OpenTracker software without any issue.
Can you update the status of this issue?

closed 2017-04-05 by ppescher