geolink/opentracker #68: Open Tracker - failed auto-baudrate error

haroon-rehman, 2017-04-10:
Hi,
After couple of days of working my open tracker gives me only one message on serial port "ailed auto-baudrate error" and after 4 or 5 retry attempts it stops gsm connection. What can cause this and how to fix it. I have attached image of what i see on serial monitor.

Thanks for any help offered.

Regards,
Haroon
[image]

haroon-rehman, 2017-04-10:
[image]

ppescher, 2017-04-10:
Try to add the following code after https://github.com/geolink/opentracker/blob/master/OpenTracker/gsm.ino#L62
`debug_print("read status");`

Just before the line that starts with "return".

Your code should look like:
```
#if MODEM_UG96
  // inverted status signal
  debug_print("read status");
  return digitalRead(PIN_STATUS_GSM) != HIGH;
#else
```

Load the modified sketch and check if you can see the message "read status" in the serial log.

haroon-rehman, 2017-04-11:
Hi Paolo,

Yes I can see the 'read status' message in the logs.

[cid:eebaf623-b8b4-49b1-87e8-c9e80ea20caf]

Regards,

Haroon

Orion Systems

haroon-rehman, 2017-04-11:
[image]

ppescher, 2017-04-11:
Stupid question, but I must ask: are you sure you have the 3G modem on your board?
You should see "Quectel UG96" written on the top white label.

If this is actually an UG96 (3G) then the modem is not starting up (STATUS pin never goes high). Which is confirmed by the green LED always off.

There might be several causes to that. First, please check your power input: if the input voltage is too low or the current provided by your power source is not enough, the modem will not start.

If you have a Voltmeter and some spare male headers or solid wires, you can probe the following voltages on the expansion connector:

[image]

Please report your readings, along with your power source voltage/current rating.

haroon-rehman, 2017-04-11:
Hi Paolo,
It is a UG96 chip.
The power adapter used is 12VDC, 1500 mA. The tracker worked on it for almost 2 days before i started getting this problem
The voltages on the expansion connector are same as indicated in your pics. I have attached some images below;
[image]

[image]

[image]

[image]

ppescher, 2017-04-11:
Voltages are correct, also power supply rating is fine.

Could you try to power on the device without the SIM card?
And check the serial log and the green LED.

haroon-rehman, 2017-04-11:
Hi mate,
Its still the same. No green light. See the image below.

[image]

Codematic, 2017-08-21:
Any final solution for this? I have the same issue which appeared after first flash try.

ppescher, 2017-08-22:
Here the modem did not reply at all, your case is different. See https://github.com/geolink/opentracker/issues/72

closed 2017-08-22 by ppescher