geolink/opentracker #70: Wrong date for gsm set/get time

manusanilea, 2017-05-04:
Hi,
There is a problem for setting and getting the date from GSM module.
The format of `time_char` variable is "02/12/98,15:56:43" (ref to gps.ino l.151)
With the commit 9c20be6a1216bb7f2f743cf1cb42f24dcc692f9f, the day and year have been inverted, but in the gsm file, they are bad :

- `gsm_set_time` call command "`AT+CCLK=` + `time_char`", and acording to [M95 AT commands manual](http://eddywireless.com/yahoo_site_admin/assets/docs/M95_AT_Commands_Manual_V12.196112248.pdf), §14.1, the expected format is "yy/MM/dd,hh:mm:ss±zz"
- `gsm_get_time` give date/time as same format but it is copied to `time_char` wich is not same format

I suggest to change `time_char` format or to invert date and year in both functions `gsm_set_time` and `gsm_get_time`

manusanilea, 2017-05-04:
Oh, my bad, the problem has been corrected, but the comment has not.

The real date format of `time_char` is "98/12/02,15:56:43", isn't it ?

ppescher, 2017-05-04:
Could you please explain where in the gsm.ino file the wrong format is used?

The commit you refer to has fixed the date format to YY/MM/DD, which is what the AT+CCLK command expects. On the contrary, the GPS format was DDMMYY, which has been converted to the correct format.

ppescher, 2017-05-04:
Posted at the same time ;)

Yes the format has been fixed and it is YY/MM/DD,hh:mm:ss+00 in the time_char variable.

manusanilea, 2017-05-04:
Thank you for your answer ;)

closed 2017-05-04 by manusanilea