geolink/opentracker #71: Add feature for "heartbeat" -able to send location in a selected interval when engine running and other when running?

MbyMatter, 2017-06-24:
its a great feature that the ALWAYS_On flag can be set - but I was looking for both, if that can be done?
so one interval that can be set then the ignition is off - and another when the ignition is on.
could be a "heartbeat interval 3600000" - or just 0 = no heartbeat when not ignition
and still keep the interval as always when the ignition is on.

ppescher, 2017-06-26:
Yes, this would be nice to have. We'll keep it in mind for future developments.
Or you may contribute your own implementation in the meanwhile. ;-)

MbyMatter, 2017-07-01:
Good, just wanted to check up if that was in the casting in the near future, I might give it a go looking at the code and add it :)