geolink/opentracker #72: Can not connect to remote server:  updates.geolink.io Modem Reply: +CEER: "No report available" Modem Reply: +CME ERROR: 4

Codematic, 2017-08-21:
I have serious problems to program OpenTracker. I have tried several source codes (3.1.1, 3.1.2 and 3.1.3) but non of those seems to work properly. Also tried to consider to have correct board support package for each of those source code packages (1.0.3 vs. 1.0.4). But that did not help either. Hope some of you have quick solution to this issue.

The board should be 3G version (HW rev 2.4C).
[image]

To make sure, I have used this setup with factory firmware more than 2 months and also same power supply were just fine for couple of hours with factory firmware.
[image]

Could there be something I have missed while creating tracker.h? (something to define more?)

Also same SIM card is used before without problems.

Is there something else I could try to modify to sort out this issue?

Btw, should all of those source code packages work with this board?

Here is log of issues with the 3.1.2 package:
```
setup() started
settings_load() started
settings_load(): First run flag:
255
settings_load(): setting defaults from config
settings_load(): set config.interval:
10000
internet
settings_save() started
settings_save() finished
settings_load(): config.interval:
10000
settings_load(): config.interval_send:
1
settings_load(): config.powersave:
0
settings_load() finished
store_get_index() started
store_get_index(): Not found, initialize log position:
1024
store_get_index() ended
gps_setup() started
gps_on() started
gps_on() completed
$PMTK011,MTKGPS*08
$PMTK010,001*2E
$PMTK011,MTKGPS*08
$PMTK010,002*2D
gps_setup() completed
gsm_setup() started
gsm_off() started
gsm_off() finished
gsm_on() started
gsm_send_at() started
gsm_send_at() completed
0
gsm_on(): failed auto-baudrate
gsm_off() started
Modem Reply:

RDY

gsm_off() finished
gsm_off() started
gsm_off() finished
gsm_on(): try again
1
gsm_send_at() started
Modem Reply:

OK

[... 2093 lines cut ...]
#define PACKET_SIZE_DELIVERY 3000     //in case modem has this number of bytes undelivered, wait till sending new data (3000 bytes default, max sending TCP buffer is 7300)

#define CONNECT_RETRY 3               //how many times to retry connecting to remote server
#define CONNECT_TIMEOUT 15000         //how much time to wait for each connection attempt

#define STORAGE 1                     //save logs in flash storage
#define STORAGE_FIRST_RUN_PAGE 0      //flash index to store flag for first run - very first page
#define STORAGE_CONFIG_PAGE 1         //flash index to store configuration - just after FIRST_RUN page
#define STORAGE_CONFIG_ADDON 512      //flash index to store addon configuration
#define STORAGE_DATA_START 1024       //starting flash index to store logs (first bytes reserved for configuration)
#define STORAGE_DATA_END 131071       //the last possible flash index (max 131072, one last byte must be reserved for marker)
//#define STORAGE_DATA_END 2048       //the last possible flash index (max 131072, one last byte must be reserved for marker)

#define STORAGE_MAX_SEND_OLD  5       //maximum number of old data to send from log

// List of APN configurations to scan at power-on/reset
// (undefine to use the default APN only)
//#define KNOWN_APN_LIST \
  //KNOWN_APN("example.apn", "user", "pass", "ISP name", NULL) \
  //RECORD FORMAT (ISP and NULL are not used) - INSERT ABOVE
  //KNOWN_APN("APN", "USERNAME", "PASSWORD", "ISP", NULL) \

#define KNOWN_APN_SCAN_MODE 1       // 0=check GPRS context only, 2=attempt only server connection, 1=use both in sequence
#define KNOWN_APN_SCAN_USE_RESET 0  // 1=reset modem to change APN, 0=change APN without reset
```

ppescher, 2017-08-22:
Looking at your log I can see the modem is working, but AT command echo has been turned off.

Recent firmware versions perform auto-baudrate using the "ATE1" command string, that should turn command echo back on, see for instance: https://github.com/geolink/opentracker/blob/55f2d320504bc305fd885f1ff1d5d4da5c12f680/OpenTracker/gsm.ino#L381

However, in your log the modem reply is only "OK", without command echo. Please check that your firmware does send the "ATE1" command, because old versions only used "AT".

Codematic, 2017-08-24:
Thank you for your prompt reply.

I investigated the gsm.ino files I tried with no luck and none of them put the echo on (ATE1).

After realising this I cloned the latest repository where that is on and found out that "RDY" is replied back to cpu from modem. According to "Quectel_WCDMA_UGxx_AT_Commands_Manual_V1.7.pdf" it is not right (should return OK or ATE1).

This is very strange since I managed to run the OpenTracker with original software without any problems (I even took a log for that and there seems to be ATE1 coming back properly).

With this issue, I even tried an other power source, but this is now something else than power source issue. (In other words: No help from that change.)

Please see the log with the latest code from the up to date repository:
```
setup() started
settings_load() started
settings_load(): First run flag:
1
settings_load(): no a first run, loading settings.
settings_load(): config.interval:
5000
settings_load(): config.interval_send:
4
settings_load(): config.powersave:
1
settings_load(): config.debug:
1
settings_load(): config.queclocator:
0
settings_load() finished
store_get_index() started
store_get_index(): Not found, initialize log position:
1024
store_get_index() ended
gps_setup() started
gps_on() started
gps_on() completed
$PMTK011,MTKGPS*08
$PMTK010,001*2E
$PMTK011,MTKGPS*08
$PMTK010,002*2D
gps_setup() completed
gsm_setup() started
gsm_off() started
gsm_off() finished
gsm_on() started
gsm_send_at() started
gsm_send_at() completed
0
gsm_on(): failed auto-baudrate
gsm_off() started
Modem Reply:

RDY

gsm_off() finished
gsm_off() started
gsm_off() finished
gsm_on(): try again
1
gsm_send_at() started
gsm_send_at() completed
0
gsm_on(): failed auto-baudrate
gsm_off() started
Modem Reply:

RDY

gsm_off() finished
gsm_off() started
gsm_off() finished
gsm_on(): try again
2
gsm_send_at() started
gsm_send_at() completed
0
gsm_on(): failed auto-baudrate
gsm_off() started
Modem Reply:

RDY

gsm_off() finished
gsm_off() started
gsm_off() finished
gsm_on(): try again
3
gsm_send_at() started
gsm_send_at() completed
0
gsm_on(): failed auto-baudrate
gsm_off() started
Modem Reply:

RDY

gsm_off() finished
gsm_off() started
gsm_off() finished
gsm_on(): try again
4
gsm_send_at() started
gsm_send_at() completed
0
gsm_on(): failed auto-baudrate
Warning: timed out waiting for last modem reply
Reply failure count:
1
Warning: timed out waiting for last modem reply
Reply failure count:
2
gsm_on() finished
gsm_set_pin() started
Warning: timed out waiting for last modem reply
Reply failure count:
3
Warning: timed out waiting for last modem reply
Reply failure count:
4
Warning: timed out waiting for last modem reply
Reply failure count:
5
Warning: timed out waiting for last modem reply
Reply failure count:
6
Warning: timed out waiting for last modem reply
Reply failure count:
7
gsm_set_pin() completed
gsm_get_modem_status() started
Warning: timed out waiting for last modem reply
Reply failure count:
8
gsm_get_modem_status() returned: 
-1
gsm_get_modem_status() started
Warning: timed out waiting for last modem reply
Reply failure count:
9
gsm_get_modem_status() returned: 
-1
gsm_get_modem_status() started
Warning: timed out waiting for last modem reply
Reply failure count:
10
reboot() started
gps_off() started
gps_off() completed
gsm_off() started
```

The OpenTracker will loop this constantly..

gsm.ino "gsm_send_at()" looks following now:

```
int gsm_send_at() {
  debug_print(F("gsm_send_at() started"));

  int ret = 0;
  for (int k=0; k<5; ++k) {
    gsm_port.print("ATE1\r");
    status_delay(50);
  
    gsm_get_reply(1);
    ret = (strstr(modem_reply, "ATE1") != NULL)
      && (strstr(modem_reply, "OK") != NULL);
    if (ret) break;

    status_delay(1000);
  }
  debug_print(F("gsm_send_at() completed"));
  debug_print(ret);
  return ret;
}
```

Any more ideas how to solve this issue? Could there be some setting in modem which could be resetted somehow?

Do you have any similar HW rev 2.4C boards available there to test the same? Could this same issue be repeated(?).

Hope you could reply soon. Thank you in advance.

ppescher, 2017-08-25:
The RDY message is a kind of "URC" (unsolicited result code) which is sent by the modem to notify about particular events (in this case the modem is ready to accept commands).
The problem is that the modem appears to be "ready" (RDY) after all attempts to receive a reply.

Try to `#define DEBUG 10` in your Tracker.h
That will start a debug terminal after modem initialization (which will fail after 1 minute in your case) and you will be able to manually send AT commands to the modem.
Things you can try to type:
AT
ATI
ATE1
Until you see an echo of what you type on the serial monitor, just before the modem reply. 

If you can get sensible replies from the modem and enable the echo, then it might be a timing issue. The code does not wait long enough for the modem to become ready and start replying normally. In that case it should be enough to modify gsm_send_at() to retry 10 times for example, instead of 5 (every attempt takes roughly 1 second).

Codematic, 2017-08-25:
Well, thank you once again for your support. The initial issue were not to define `MODEM_UG96` at all in tracker.h. After that, I got it working somehow, but still connecting to server is not working. It seems to be right here.

The key error message is `+CEER: "No report available"` as shown below.

```
Data accumulated:
4
gsm_send_at() started
Modem Reply:
ATE1

OK

gsm_send_at() completed
1
gsm_disconnect() started
Modem Line:
AT+QICLOSE=0

Modem Line:
OK

gsm_disconnect() completed
gsm_connect() started
gsm_get_modem_status() started
Modem Reply:
AT+CPAS

Modem Reply:
+CPAS: 0

Modem Line:

Modem Line:
OK

gsm_get_modem_status() returned: 
0
gsm_get_connection_status() started
Modem Line:
AT+QISTATE=1,0

Modem Line:
OK

Modem Line:
AT+CGACT?

Modem Line:
+CGACT: 1,0

Modem Line:

Modem Line:
OK

Modem Reply:
AT+CGACT?

+CGACT: 1,0

OK

gsm_get_connection_status() returned:
-2
Modem Line:
AT+QIDEACT=1

Modem Line:
OK

Modem Line:
AT+QIACT=1

Modem Line:

Modem Line:
OK

Modem Line:
AT+QIDNSCFG=1,"8.8.8.8"

Modem Line:
OK

Connecting to remote server...
0
Modem Line:
AT+QIOPEN=1,0,"TCP","updates.geolink.io","80"

Modem Line:
ERROR

Can not connect to remote server: 
updates.geolink.io
Modem Line:
AT+CEER

Modem Line:
+CEER: "No report available"

Modem Line:

Modem Line:
OK

gsm_get_modem_status() started
Modem Reply:
AT+CPAS

Modem Reply:
+CPAS: 0

Modem Line:

Modem Line:
OK
```

In gsm.ino it is about this function :
```
int gsm_connect() {
  int ret = 0;

  debug_print(F("gsm_connect() started"));

  //try to connect multiple times
  for(int i=0;i<CONNECT_RETRY;i++) {
    // connect only when modem is ready
    if (gsm_get_modem_status() == 0) {
      // check if connected from previous attempts
      int ipstat = gsm_get_connection_status();  

      if (ipstat > 1) {
        //close connection, if previous attempts failed
        gsm_port.print(AT_CLOSE);
        gsm_wait_for_reply(MODEM_UG96,0);
        ipstat = 0;
      }
      if (ipstat < 0) {
        //deactivate required
        gsm_port.print(AT_DEACTIVATE);
        gsm_wait_for_reply(MODEM_UG96,0);
        ipstat = 0;

#if MODEM_UG96
        gsm_port.print(AT_ACTIVATE);
        gsm_wait_for_reply(1,0);
        
        gsm_port.print(AT_CONFIGDNS "\"8.8.8.8\"");
        gsm_port.print("\r");
      
        gsm_wait_for_reply(1,0);
#endif
      }
      if (ipstat == 0) {
        debug_print(F("Connecting to remote server..."));
        debug_print(i);
    
        //open socket connection to remote host
        //opening connection
        gsm_port.print(AT_OPEN "\"");
        gsm_port.print(PROTO);
        gsm_port.print("\",\"");
        gsm_port.print(HOSTNAME);
        gsm_port.print("\",");
#if MODEM_M95
        gsm_port.print("\"");
#endif
        gsm_port.print(HTTP_PORT);
#if MODEM_M95
        gsm_port.print("\"");
#endif
        gsm_port.print("\r");
    
        gsm_wait_for_reply(1, 0); // OK sent first

        long timer = millis();
        if(strstr(modem_reply, "OK")==NULL)
          ipstat = 0;
        else
        do {
          gsm_get_reply(1);

#if MODEM_UG96
          char *tmp = strstr(modem_reply, "+QIOPEN: 0,");
          if(tmp!=NULL) {
            tmp += strlen("+QIOPEN: 0,");
            if (atoi(tmp)==0)
              ipstat = 1;
            else
              ipstat = 0;
            break;
          }
#else
          if(strstr(modem_reply, "CONNECT OK")!=NULL) {
            ipstat = 1;
            break;
          }
          if(strstr(modem_reply, "CONNECT FAIL")!=NULL ||
            strstr(modem_reply, "ERROR")!=NULL) {
            ipstat = 0;
            break;
          }
#endif
          addon_delay(100);
        } while (millis() - timer < CONNECT_TIMEOUT);
      }
      
      if(ipstat == 1) {
        debug_print(F("Connected to remote server: "));
        debug_print(HOSTNAME);
  
        ret = 1;
        break;
      } else {
        debug_print(F("Can not connect to remote server: "));
        debug_print(HOSTNAME);
        // debug only:
        gsm_port.print("AT+CEER\r");
        gsm_wait_for_reply(1,0);
      }
    }

    addon_delay(2000); // wait 2s before retrying
  }
  debug_print(F("gsm_connect() completed"));
  return ret;
}
```

Any ideas what to test next?

Btw, with the `#define DEBUG 10` the code did not reached the testing phase at all since the MODEM_96 were not defined (it just rebooted earlier).

Thanks.

ppescher, 2017-08-25:
Ok, you don't have to define MODEM_UG96 manually. It is defined automatically when you select the correct board hardware in the Arduino IDE: in the menu Tools > Board Version you should set "OpenTracker 2.5 (3G)" (the board revision number is a mistake, it's ok for 2.4C and will be corrected in the next release).

Please make sure you have configured the APN correctly for your SIM card. I don't see the initial diagnostic messages in your log.

Codematic, 2017-08-25:
Finally I got it working.

I added these two lines into the gsm.ino :
[image]

Got error stating: `+QIGETERROR: 552, invalid parameters`

The real pain were these preprocessor IF-statements:
[image]

UG96 does not like parenthesis around the port number.

Summary to solve this issue with UG96 modem were just to define those two preprocessor parameters to the tracker.h:
```
#define MODEM_M95 0
#define MODEM_UG96 1
```
***************************
Just noted Paolo's last reply...

Tested the board definition and yes indeed, it will clear the same.

Hmm, such a small issue (the board number 2.5 vs. 2.4C) but causes many hour of investigation.

Anyway Paolo, thank you for your constant and prompt support. I will close this for now.

closed 2017-08-25 by Codematic

ppescher, 2017-08-25:
FYI - I've just released board support package [v1.0.5](https://github.com/geolink/opentracker-arduino-board/releases/tag/v1.0.5) (you can get it via Arduino IDE Board Manager) and firmware package [v3.2.0](https://github.com/geolink/opentracker/releases/tag/v3.2.0)

I know this comes just a bit too late for you and I'm sorry for the inconvenience.
Our release policy so far has been to follow our board production schedule, unless immediate action to fix serious bugs is deemed necessary. This time we had some delay due to summer closing. Thanks for understanding.