geolink/opentracker #79: geolink.io: trips not finalized when car stops

Chrescht, 2019-01-31:
My trip from this morning was got connected to some older trip from two days ago. That's not good.
Is there a way to define when to start a new trip? Something like setting the off time interval.

I got many undefined points: 
Speed: undefined km/h
Altitude: undefined m
Heading: undefined

The ignition cable is connected. Do you have an idea what the issue here could be?

m4rkw, 2019-01-31:
Hi @Chrescht I don't speak for Geolink but I think stuff like this (ie problems with the service you're paying for) should probably be dealt with by their support rather than by opening github issues. Try emailing support@geolink.io

Chrescht, 2019-02-13:
Which of the 16 OBD2 pins is the VDET (IGN) one to determine if the car is running? I can't seem to find this information anywhere.

geolink support: they are flooded by emails apparently so you won't get an answer anytime soon.

Chrescht, 2019-08-10:
@m4rkw thanks for your reply, I did ask them, not much came back though.
I can answer my car running question now: at least for VW cars it's pin 1 on the OBD2, (12V when key in the last positions, else 0V).

At the moment the trip A->B gets saved 3 times for some reason, and the trip B->A is missing.. I'll have to do some debugging.