geolink/opentracker #80: Can I connect to a HTTPS server?

LBreda, 2019-06-25:
Can I connect to a HTTPS server without getting only the "Moved permanently" error?

ppescher, 2019-06-26:
Only HTTP has been implemented in stock OpenTracker firmware, but the modem does support HTTPS via dedicated AT commands.

LBreda, 2019-06-26:
It's good enough for me. I checked the docs of the UG96 chip, and I didn't find anything useful. Could you give me some hint?

ppescher, 2019-06-26:
What you need is the document "Quectel WCDMA UGxx TCPIP AT Commands Manual", which is available from the manufacturer after registration (See <https://www.quectel.com/support/downloads/ATCommandsManual.htm?keys=wcdma>).

LBreda, 2019-06-26:
Thanks, I was reading a more general AT commands manual.

closed 2019-06-26 by LBreda

ppescher, 2019-06-26:
Actually, there are other manuals for HTTPS but they are included in the list above:
"Quectel WCDMA UGxx SSL AT Commands Manual"
"Quectel WCDMA UGxx HTTP AT Commands Manual"

LBreda, 2019-06-26:
Yep, found them. I'm having some trouble adapting the ack mechanism, tho.